uf-mil/mechanical #15: [UAV replenishment] Can pads

amcaleer271, 2024-09-03:
We need to build two floating pads that the drone can pick up the cans from. The pad needs to be 6ft by 6ft, and raised 0.3m from the surface of the water. There is also a decal that needs to be applied to the top face of the pads. This can be found on the task ideas powerpoint linked here: https://robonation.org/app/uploads/sites/2/2024/03/Task-Ideas-Primer_RobotX-2024.pdf.

WillMacGuire, 2024-09-09:
I did some math related to the buoyancy of this platform and materials to use. I figured we can make the platform out of 4 2x4s that are about 70.5 in long. The pad iself can be 0.5 in plywood. I found this designs mass in fusionto determine the buoancy force needed to keep the platform stable and saw we needed about 2300 in^3 of material underwater. This can be achieved with 4 legs with about 4 inches underwater each, with a 12 in by 12 in square cross sections. This means that a legs would need to be about 15.5 inches long. There could be some stability issues that may require some kind of ballast and I can look into designs with that in the future

adamhamdan8, 2024-09-09:
@WillMacGuire Hey will, great work, just try and add screenshots of any fusion work and math that you do so that we have it for future reference and what not!

adamhamdan8, 2024-09-20:
@WillMacGuire Do you have any updates?

WillMacGuire, 2024-09-21:
today I worked on trying to get a pvc solution to keep the pads elevated and allowing for variable amounts of buoyant material (likely pool noodles) to be added. The current idea has 2 pvc squares per pad to deal with outward loads and the pad will rest on top. The squares will be glued together but held together with standoffs that can be removed for ease of transport. I talked with gabe a little bit about ideas to waterproof the wood but it is unclear how needed that will be. CAD will be made tomorrow of the idea.

WillMacGuire, 2024-09-22:
[image]
Today I finished the simplified CAD model of the landing pads. The design mirrors the description from yesterday and uses 2 inch pvc pipes to support the pad. The bottom horizontal pipes are meant to be wrapped in pool noodles ziptied on to control buoyancy. I am about to conduct some calculations to see what volume of pool noodles are needed.

WillMacGuire, 2024-09-22:
[image]

I did some math for finding out how many large pool noodles (2.5" OD, 1" ID, 52" L) would be needed to support the current design. The assumption about mass is likely innacurate and off by up to 30 or 40% so it makes sense to get about 28 noodles to me. This number could be lowered if we are able to make the lower horizontal sections waterproof so their volume is used to generate buoyancy. This may be possible with some kind of rubber cement at the joints and I will do more research on this tomorrow.

WillMacGuire, 2024-09-23:
[image]

If we are able to make the horizontal sections whaterproof that would decrerase the volume from noodles needed by around 1300 cubic inches. This means that we might be able to get away with not stapping pool noodles across the middle because we only need to displace another 34 inches. This is within the margain of error for the weight and buoyancy needed so its not worth going into much detail about what else we need to add as the exact volume needed should be determined expirementally but this gives us a good starting place. I am now looking into ways to seal the pipes.

WillMacGuire, 2024-09-23:
I found a glue compound at lowes that should work at creating a waterproof seal. Tomorrow I am going to talk with adam about the parts we need and hopefully coordinate a plan to get them mostly from lowes and with noodles from amazon

WillMacGuire, 2024-09-29:
Adam clarified that this would be on land on tuesday so the idea has been modified to suit these requirments. Building should be started this next week assuming the hurricane didnt push back buying the plywood sheets later. I plan to move to other tasks for now

WillMacGuire, 2024-09-30:
Tomorrow I am going to come into the lab to check what kinds of paint we have. To make the landing pads we need to have onyx black for the rings and sidewalk grey (https://www.benjaminmoore.com/en-us/paint-colors/color/2133-60/sidewalk-gray) for the background. Its possible we might be able to get away with some slightely different coloring so I am going to talk to programmers tomorrow to get that clarified. 

Also to make the circle diameters the correct length I am planning to put a nail in the center of the circle. Then attaching a spring around the nail which is connected to the paint either in a can or bursh. The string is length of the radius and can be walked around the board to get perfect circles which will be important for the machine learning algorithm.

adamhamdan8, 2024-10-01:
> Tomorrow I am going to come into the lab to check what kinds of paint we have. To make the landing pads we need to have onyx black for the rings and sidewalk grey [...]

We will likely end up printing this out of laminated paper, I just need to talk to Target Copy and get that figured out later today. Lowe's unfortunately was missing lots of things, so we have to put a hold on that!

WillMacGuire, 2024-10-13:
[image]
I worked with Ryan on tuesday right before the hurricane for most of the day to make the designs for the can landing pad. We made combined two peices of paper before painting the concentric circles. For the design at the center we used a stencil that we printed out to ensure the design matched what it will really be. We made a small mistake in that the stencil was a bit too small by about 3 inches. We talked with programmers about this and they said its fine as long as the shape looks right as the height is monitered by lidar. The stencil works when the paper is covered with tape to stop paint bleeding through. For other pads though the plan is to source stencils from companies as they are several feet large.

adamhamdan8, 2024-10-14:
I think we are going to just can this task (haha lol) because we decided for sure not to fly the drone over water.

closed 2024-10-14 by adamhamdan8