uf-mil/mechanical #16: [Props and Obstacles] UAV Search and Report

adamhamdan8, 2024-09-06:
The UAV (autonomous drone) on NaviGator is going to be doing tasks off of the boat, and props need to be made for those tasks. For this particular task, the drone is flying between two helipads and reporting the location of two marked squares. The image is included below, but you should check the 2022 RobotX team handbook for specific details. Expect to also communicate with Adam Hamdan about purchasing the required materials and getting vinyl decals printed at Target Copy.

Deliverables for this task will be
-  Two helipads
 - Compile a BOM and get files ready for printing
- Two marked plates for identification
 - Compile a BOM and get files ready for printing
- (Possibly) Four orange traffic cones
 - Probably purchased

[image]

RyanHoburg, 2024-09-07:
I've been looking at a lot of different materials to use for the helipad because it needs to be stiff and stay flat/level despite its large 5x5 foot size. I think the most cost effective and quick solution is a 5x5 sheet of plywood and 4 spiked and angled table legs so that you can sink the legs a bit into the grass field and have a level and stable platform. I considered printing them but if they have to support a sheet of plywood I figured PETG wouldn't cut it so I found a set of metal legs on amazon for less than $30. [https://a.co/d/ecd2w0Y](url) This link is for the 12 inch legs, I think that the 12 inch or 7 inch legs would work the best.

RyanHoburg, 2024-09-07:
BOM:
- 4 orange traffic cones (stock orange)
- 1 5x5 ft sheet of plywood per helipad
- 1 set of 4 table legs per helipad (linked in above comment)

RyanHoburg, 2024-09-07:
I requested a quote from target copy asking if they can print the design of the helipad/objects big enough and on a durable material, I will add another comment when they reach back out. We can also paint the plywood but I think having it printed will keep it much more exact.

RyanHoburg, 2024-09-07:
As far as the objects go I think that they should also be made with plywood because it is the cheapest way to make a flat 5x5 foot surface. I looked for corrugated plastic sheets and other alternatives but they are much more expensive. We can also just stack all the plywood pieces together for transportation which I think would work well.

adamhamdan8, 2024-09-07:
> I requested a quote from target copy asking if they can print the design of the helipad/objects big enough and on a durable material

I have worked with them a bunch already, we can definitely get it printed on tarp with metal grommets so that we attach it to the top of the helipad. Overall great work here! I will start purchasing stuff for this early next week for manufacturing.

RyanHoburg, 2024-09-23:
[image] Price seems excessive, if someone who has ordered stuff like this before from target copy can recommend what to order it could save money

cbrxyz, 2024-09-23:
@RyanHoburg

wow, did it really take them that long to get back to you? :(

Could you use a gray tarp, and paint black circles on it? Then, all you would need is the RoboNation logo in the center, which you could either print out and stick in the center with clear tape/glue/etc., or have a transparent sticker printed and just put it in the center!

RyanHoburg, 2024-09-23:
@cbrxyz No sorry it didn't take them that long I just started working on another issue and forgot to add the quote. I was talking to Adam about it on Saturday and he said that he has used target copy for this stuff before and it usually isn't as expensive so I was gonna ask him more about it when I see him next

adamhamdan8, 2024-09-30:
I spoke with @RyanHoburg at some point last week and we are going to use laminated paper, since the pad should really only have sun exposure. Will buy that tomorrow!

RyanHoburg, 2024-10-01:
[image]
[image]
[image]

adamhamdan8, 2024-10-03:
> ![Screenshot 2024-10-01 152727](https://private-user-images.githubusercontent.com/180553860/372621015-bee0befd-55a1-44ce-933e-93579570ee5e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5Nzc5NzcsIm5iZiI6MTcyNzk3NzY3NywicGF0aCI6Ii8xODA1NTM4NjAvMzcyNjIxMDE1LWJlZTBiZWZkLTU1YTEtNDRjZS05MzNlLTkzNTc5NTcwZWU1ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwM1QxNzQ3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YmNmNzIxNDJkYzM1YWVkMDFjY2EyNzY2Mjk5Y2IwNzI5OGNjZjkxMGMxZDMxMmYwNDE5YjQwMGZiNGI3NTliJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7PWp9THPfyP4RavHSbhR6pVkdTOO2Tz8woBq1hGt654) ![Screenshot 2024-10-01 152744](https://private-user-images.githubusercontent.com/180553860/372621020-afc02396-c86f-4c6d-8b84-96714a82168e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5Nzc5NzcsIm5iZiI6MTcyNzk3NzY3NywicGF0aCI6Ii8xODA1NTM4NjAvMzcyNjIxMDIwLWFmYzAyMzk2LWM4NmYtNGM2ZC04Yjg0LTk2NzE0YTgyMTY4ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwM1QxNzQ3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MTE0YTM4N2FjNzIwNjU2ZTg5NWNmOTE0ZjNhNDAxYzM0ZDYyYzlkYjBmY2Y2ZWIxNTVhNDRhNTk5NjMyMTBkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.pDiNHtjo1v8msji63X8gewmPKm2H1__fWdf3nz6L_94) ![Screenshot 2024-10-01 152800](https://private-user-images.githubusercontent.com/180553860/372621022-a8ba7d1f-747d-4d8b-b299-e44e024a2959.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5Nzc5NzcsIm5iZiI6MTcyNzk3NzY3NywicGF0aCI6Ii8xODA1NTM4NjAvMzcyNjIxMDIyLWE4YmE3ZDFmLTc0N2QtNGQ4Yi1iMjk5LWU0NGUwMjRhMjk1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwM1QxNzQ3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNWNjNmYyNTRkMDZmZDA1ZThhMzdjMTUxMjhhY2FiMTg2OGU3NTk4ZjY1N2EzMTk3ZWQ0ZjA3YTNhN2E0MzllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.L0HCZFWnlKBhloxUX0gho2q25fJXAJCQiAl4dz_2bZs)

We're going to have to paint this as well as we can to fit the needs of the software team. A grey tarp with a similar color, and black paint with a similar color. The color names are located in the handbook and I will include @cbrxyz 's requests for the tarp.
[image]

RyanHoburg, 2024-10-03:
Required materials put in the order form, Diego and I are ready to get on it as soon as we have the paint and tarp

DiegoDom23, 2024-10-07:
We have a plan of how we are going to go about painting the tarp and getting the designs on there. We expect to be able to complete this week if materials are found and weather allows.

adamhamdan8, 2024-10-07:
@RyanHoburg @DiegoDom23, @lorantdomokos went to Lowe's today and brought the stuff. The grey tarp is sitting in the lab, and we have black paint and spray paint for the pattern to be put on the tarp for the landing pad. Whenever you guys get a chance, but I know that the lab will be closed tomorrow which sucks. Maybe get a stencil or something for the middle robot guy, and just try and be as accurate as you can with the circle and the centering.

One last note, the tarp is 40 inches wide so try and combine them together. Maybe on a piece of plywood but that would be really annoying to move.

RyanHoburg, 2024-10-08:
[image]
@WillMacGuire and I got it

RyanHoburg, 2024-10-08:
We need a smaller paint brush to finish some of the finer details but that will take 10 seconds once we have the brush

adamhamdan8, 2024-10-17:
I saw that you guys had finished with the smaller details, and I am waiting on the pcard to get the smaller two symbols printed on laminated paper. Does that seem fine with you guys? Also could you get the sizes for those just somewhere in here

RyanHoburg, 2024-10-21:
Finished details on landing pad, decided not to staple it to anything so it's easy to transport. This task is done

adamhamdan8, 2024-10-23:
> Finished details on landing pad, decided not to staple it to anything so it's easy to transport. This task is done

Great I'm going to close the issue now! We also could use the other two smaller ones painted but I don't think the drone will be ready to test those things.

RyanHoburg, 2024-11-03:
Painted one of the task tarps [image]