uf-mil/mechanical #17: Thruster Shrouds

amcaleer271, 2024-09-09:
The thrusters on NaviGator must be shrouded to ensure safety during the competition (Handbook, pg. 34, item 5). 

These shrouds should prevent large items from entering the thrusters, but they do not need to completely cover them.

adamhamdan8, 2024-09-10:
Previous years they used something silly like a wire basket, so the requirement is not super strict, but preferably something nice

csachs32, 2024-09-17:
I can work on this

csachs32, 2024-09-19:
Designed two parts that connect together via 1/4-20 bolts that shroud the thruster. The two parts and an assembly with a thruster can be located in the "Thruster Shroud" Folder in "Navigator AMS". A picture of the design is attached.

adamhamdan8, 2024-09-20:
Taking a quick look now, that looks good! I can't really think of a more elegant solution for right now, and this would be quick to do. We can print a prototype and see what we like and what we don't like!

csachs32, 2024-09-22:
Uploaded a prototype to the 3D printer and am waiting for it to print/ next time I go to the lab to test it out.

csachs32, 2024-09-24:
Printed the first half of the shroud and will start the second half print today.

csachs32, 2024-10-05:
Printed the first prototype of the shrouds, but they do not fit around the thrusters correctly. Need to make changes to the design.

csachs32, 2024-10-05:
Made changes to the design, but now the shrouds cannot print on the 3D printer because they are too big. Currently troubleshooting this.[image]

csachs32, 2024-10-14:
We might have to print these at the rapid proto lab. Waiting on the leads to discuss this dilemma.

csachs32, 2024-10-15:
Split each side of the thruster in half so that there are now 4 parts for each shroud. Sent the parts to the print queue, and once printed, will weld each part together to test the prototype.

adamhamdan8, 2024-10-23:
Lots of things are in the print queue, so this will be on hold for a bit!

csachs32, 2024-11-04:
Printed final shrouds for each thruster (and an extra shroud), and welded one thruster together to mount to NaviGator.

adamhamdan8, 2024-11-12:
We had to macgyver like four different solutions. The current one is still on navigator for all four thrusters. Created by @adamhamdan8 @amcaleer271 @space-cat-explorer Should be sufficient for the future

closed 2024-11-12 by adamhamdan8