uf-mil/mechanical #19: Antenna Securing Method

amcaleer271, 2024-09-13:
The main antenna for NaviGator (large cylinder on top near the front, which looks like a piece of PVC) needs to be secured during transport and held upright during deployment. 

The antenna should be held down securely during transport, but able to be deployed reasonably quickly, ideally without tools. During testing, the antenna should securely be held upright.

space-cat-explorer, 2024-09-23:
Didn't get a great start on this, and won't be able to focus on it a ton this week because of exams and the career fair. I went to cimar and took pictures of the antenna to try and brainstorm.

space-cat-explorer, 2024-09-30:
I went to testing last week but couldn't go this week due to a big exam week coming up. I have some ideas for what I am going to do for this design, I just need to go to cimar and look at what holes are available to secure brackets onto.

space-cat-explorer, 2024-10-14:
![image](https://github.com/user-attachments/assets/834dda4e-22e8-4fea-bcb5-55a55c88c40f

space-cat-explorer, 2024-10-14:
[image]

space-cat-explorer, 2024-10-14:
Thinking about using a collapsible 90 degree bracket for securing the antenna, just need to go to CIMAR to look at where it could be secured so that I can add one to the order form and possibly 3D print any necessary mounts for it.

adamhamdan8, 2024-10-23:
> Thinking about using a collapsible 90 degree bracket for securing the antenna, just need to go to CIMAR to look at where it could be secured so that I can add one to the order form and possibly 3D print any necessary mounts for it.

Hey Sophie keep in mind we only have until the end of next week before the competition!

adamhamdan8, 2024-10-23:
Fine by me, we can just close the task. Thanks Sophie!

closed 2024-10-23 by adamhamdan8