uf-mil/mechanical #20: Calibrate and Maintenance 3D Printer

amcaleer271, 2024-10-03:
The printer is starting to print less good. It is supposed to be maintenanced frequently and we have never done this.

WillMacGuire, 2024-10-14:
I can come in tomorrow to do some work on this

WillMacGuire, 2024-10-14:
Here are the links that the error qr codes lead to:

https://wiki.bambulab.com/en/general/lead-screws-lubrication

https://wiki.bambulab.com/en/general/carbon-rods-clearance

WillMacGuire, 2024-10-14:
The carbon rods clearance issue says that the materials needed are an Isopropyl alcohol spray or IPA and a microfiber cloth. For the lead screws lubrication https://www.oilbx.com/food-machinery-grease/bx-300-f(df)-series-food-grade-high-temperature-fluorine-containing-grease.html is a reccomended grease. It also says that Super Lube 92003 Silicone Lubricating Grease with PTFE or Lucas Oil 10533 White Lithium Grease should work similarly. 

Aditinally it would make sense to make a system to track what maintenance was done and when on the printer, the lead screws are meant to be lubricated every 3 months so it shouldnt be a super involved process to setup.

WillMacGuire, 2024-10-21:
just sent out the order request to get the super lube from the link, going to clean the carbon rods tomorrow

WillMacGuire, 2024-10-22:
I spent a couple hours yesterday trying to clean the carbon rods. They no longer have any build up on them that shows up when you try to clean it further but the printer is saying to do maintenance on them still.

adamhamdan8, 2024-10-23:
Just sent the email to order the lube to professor Schwartz

amcaleer271, 2025-05-01:
Sophie and I did this yesterday:
Cleaned carbon rods
Greased Z axis screws
Calibrated printer

closed 2025-05-01 by amcaleer271