uf-mil/mechanical #23: New Electronics Box

amcaleer271, 2024-11-22:
Problem:
The luggage carrier containing the electronics for NaviGator is difficult to access and work with, and the electronics inside are disorganized and at times fly around freely. 

Requirements:
* New electronics box is easy to access and maintain
* Modular assembly to support multiple electronics boxes and future expansion
* Electronics boxes should be easy to disconnect and remove so they can be worked on in the lab or other locations.
* All electronics components should be mounted securely and labeled.

amcaleer271, 2024-11-22:
Current idea is to build an 80-20 superstructure to support the new electronics boxes, and also allow mounting of sensors and other structures on top of NaviGator

amcaleer271, 2025-01-28:
We have chosen some off-brand boxes for the electronics (linked below). The plan is to use one of the larger boxes for the high power box, and two of the smaller boxes for the MIL boards box and the computer box. We will then have one spare box for future expansion. 

We still need to figure out a cooling solution for each box. Schwartz wants a cooling loop to go through all boxes but we don't know how this is feasible.

space-cat-explorer, 2025-01-29:
Link to BOM for stage 0: https://uflorida-my.sharepoint.com/:x:/g/personal/ems_ufl_edu/EWdh7Iba5eZOiNglALGu7B0BinelZChBztsa4OkIl0RQDA?e=2hoKid

Stage 1: begin assembling boxes (secure polycarbonate sheets to boxes with head set inserts inside 3D printer standoffs) and start converting electronics over. Also begin assembly of 80-20 frame on boat (need to remove old 80-20 from new boat and figure out how to do latching system for the boxes).

space-cat-explorer, 2025-01-29:
@nonagaga is working on cooling for each box. We plan to keep the existing water cooling loop in the computer box and implement fans into the other boxes.

amcaleer271, 2025-01-29:
We will have to modify the computer box so that the radiator for the water cooling loop is attached to the pelican box. I'm going to wait until the boxes come in before working too hard on a latching system for the boxes as I'm not able to find a reliable CAD of them anywhere.

nonagaga, 2025-01-31:
> [@nonagaga](https://github.com/nonagaga) is working on cooling for each box. We plan to keep the existing water cooling loop in the computer box and implement fans into the other boxes.

I have CAD'ed up a cooling fan assembly for the smaller pelican cases. Next steps involve designing an external splash guard and rain cover to minimize water in the cases. The specific fans used are generic PC fans, 120mm x 120mm, and they're linked below:

https://www.amazon.com/Airflow-Computer-120x120x25mm-4-72inch-Brushless/dp/B09N6WQ616/

nonagaga, 2025-02-07:
Fans have arrived in lab and were tested for back EMF yesterday with Jack's help. When the fan turned off, we did not see any back EMF.

[image]

nonagaga, 2025-02-16:
Ordered new polycarb for the interior mounting plate with @space-cat-explorer. Finalizing CAD for the larger high power box.

nonagaga, 2025-04-06:
Went to DML this week to machine the iron support brackets for high power and computer box. Starting on a 3D-printed mount for the PSU, which will use some of the extra brackets as rigid supports.