uf-mil/mechanical #24: [Mounting] Lidar Mount

amcaleer271, 2024-11-22:
The current Lidar mount is large and heavy, and has a fixed mounting pattern. This makes it difficult to change and incompatible with other planned renovations of NaviGator.

Requirements:
* Maintain current position and angle of the lidar relative to the boat.
* Able to be mounted to an 80-20 aluminum extrusion using 1/4-20 hardware.
* Extremely rigid, the mount cannot move due to movement of NaviGator
* Easy to attach and detach either the entire mount or the lidar. The lidar cannot stay on NaviGator during transit.

amcaleer271, 2025-01-22:
Assigned to Victer, tentative deadline of design by 2/13

VicterQ, 2025-01-27:
Researching on how lidars work, what they need and how best to utilize them, how best to mount to 8020 and ways to reduce vibrations, what types of vibrations and environmental hazards likely to see, and attachment methods for easy disassembly or folding mechanism.

space-cat-explorer, 2025-01-30:
@raffimadalian10 I am adding you to this because we accidentally assigned both you and Victer to this task separately. You guys both have the same idea for how to mount it onto 80-20 so just add any info you have to this issue instead of the antenna mounting issue.

raffimadalian10, 2025-01-30:
- Able to mount existing lidar holding apparatus to two parallel 8-20 aluminum extrusions securely
- Need 4x single nut with center hole T-slotted framing fasteners and 4x 1/4"-20 Thread Size, 1/2" Long

raffimadalian10, 2025-01-30:
@VicterQ Im working on getting the mount into the full assembly right now -- the subassembly is under sensors and lidar mount

VicterQ, 2025-01-30:
@raffimadalian10 Dude that's awesome! But I did have a concern with the possible vibrations. I'm not completely sure how much vibrations there might be so I want to drill through the 80/20 to fit a larger (lock) nut, as the T slot nuts only allow for like 2 threads of engagement and can be rattled off with enough vibrations and time. What do you think about that?

raffimadalian10, 2025-01-31:
Sounds great, good idea. Yeah when I tested out fitting it on a piece of 80/20 the 1/2" screw was the largest you could go without hitting the bottom of the 80/20 but it barely fit through the mount plate and the T slot nut. We can still maybe use the T slot nut and just use a longer screw after drilling? Or if a larger lock nut fits in there that works too.

amcaleer271, 2025-01-31:
Another option to keep in mind is there are t-nuts with [threadlocker](https://www.mcmaster.com/products/~/t-slotted-framing-and-fittings~/t-slotted-framing-fasteners-4/t-slotted-framing-nut-type~single-with-threadlocker/) available to help reduce vibrations.

raffimadalian10, 2025-02-03:
- Subassembly for lidar mount finished. Just need to put into full assembly redesign. Going to use Heavy Duty Locking-Slotted Framing and Fittings End Feed Nut to reduce vibrations.

VicterQ, 2025-02-03:
Looked into options for mounting to 80/20 to reduce vibrations without drilling. Worked with Raffi to brainstorm and discuss options available.

amcaleer271, 2025-02-05:
I looked at the mount that is in the fusion, the way the mount is attached looks fine, but it needs to be moved further forward to avoid accidental collisions with the drone. It can be moved to the front of the top plate

raffimadalian10, 2025-02-09:
Finished fixing lidar mount subassembly - switched out current nuts for correct nuts. Fixed positioning of mount on full assembly. Ready for new task.

raffimadalian10, 2025-04-21:
Update: Will be able to get the lidar mount mounted on the boat next semester -- everything is set just waiting on the 80/20 frame. 
- Sorry, I have a lot of exams this week and haven't had much time to spend on another task. Will give final updates on current tasks and try and work on a wiki post next week.

raffimadalian10, 2025-11-05:
- Mounted the Lidar on the 80/20 frame on Navigator yesterday 

[image]
[image]

RyanHoburg, 2025-11-05:
A different 90 degree bracket for 80/20 is needed on one corner, or a shorter 1/4-20 bolt, to fully tighten it too. Mount looks great!

adamhamdan8, 2025-11-21:
lidar is mounted! yippee! Yay! ;)

closed 2025-11-21 by adamhamdan8