uf-mil/mechanical #25: [Mounting] Antenna Mounts

amcaleer271, 2024-11-22:
NaviGator has multiple antennas used to communicate with the Ubiquiti land antenna, joystick controllers, and RF Kill. Each of these antenna currently use different mounting solutions, and are difficult to change.

Requirements:
* Universal mounting system that can be attached to 80-20 aluminum extrusion using 1/4-20 hardware
* Feel free to use off-the-shelf parts to avoid 3D printing
* Each antenna must be rigidly held in place
* Either the entire mount or the antenna must be able to fold down and be securely stowed for transit.

space-cat-explorer, 2025-01-23:
Raffi

m-erez, 2025-01-24:
Adjustable flagpole mount might be an interesting option for the antenna with the thinner diameter (which appears to be 0.85"). Unfortunately, these only accomodate a 1" diameter cylinder (or less), so are not possible to use for the thicker antenna (1.75"). Searching for other solutions that can be purchased before modeling something.

Mount here: https://www.lowes.com/pd/Style-Selections-White-Metal-Flag-Pole-Mounting-Bracket/5013824073?store=2365&cm_mmc=shp-_-c-_-prd-_-sol-_-ggl-_-CRP_SHP_LIA_SOL_Online_E-F-_-5013824073-_-local-_-0-_-0&gad_source=1&gclid=EAIaIQobChMI4877mO-OiwMVmUdHAR0KOxaHEAQYAiABEgI5v_D_BwE&gclsrc=aw.ds

amcaleer271, 2025-01-24:
I like the idea of using a flagpole mount, just make sure there is either clearance or we'd be able to make a hole for the cable coming out of the antenna.

m-erez, 2025-01-28:
Attached some pictures, measurements, and notes about the smaller antenna from today.

[image]

[image]

[image]

raffimadalian10, 2025-01-28:
Lidar Antenna Mount Solution 
- Able to mount existing lidar holding apparatus to two parallel 8-20 aluminum extrusions securely 
- Need 4x single nut with center hole T-slotted framing fasteners and 4x 1/4"-20 Thread Size, 1/2" Long

space-cat-explorer, 2025-01-29:
@raffimadalian10 make sure whatever mounting system you create also includes fasteners for securing the 80-20 bars to the 80-20 frame we will be putting on the boat. Also make sure you figure out how much 80-20 you will be needing for the mount.

amcaleer271, 2025-01-29:
Before we buy any parts, have the full assembly in CAD and add it to the master assembly to make sure there isn't any interference.

m-erez, 2025-01-30:
Was doing some more research today, and this is an alternative option to the flag mount: https://www.amazon.com/Amarine-made-Marine-Antenna-Adjustable-Stainless/dp/B01E8FFUH0/ref=sr_1_1_sspa?dib=eyJ2IjoiMSJ9.i6fFRpk0xQEAX24wxPr_XiA_Y4GIZe0JDtZdEonJxlwXFLBA45ZFwTu-Tdbt25v77uJpW5PTQZI7FFf2b0ffximHGz4T8KL0W0wlZdqyvAEVc_Kg8_mn_cmpJAlCsL2vD3U-YfPhD24GGmDMriPxjxHCs8wm713NIErjezDBxkNnSVPcJlQ8WCckd09RHTNHNppChtViffbBGzyBkpeXgTculK_ppHyyGErTgEvK32Q.FHoog78Bl7SJeyfCv_2ANSlDxuf07rglJAN7yZ9Jl_c&dib_tag=se&keywords=Marine+VHF+Antenna+316+Stainless+Steel+Adjustable+Base+Mount+for+Boats&qid=1738208535&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1. These mounts are specifically designed for marine antennas. 2 problems: marine antennas have a specific male thread meant to attach to these, and no space for cables. Might be able to find an adaptor that works with our current antennas (and allows cables to come out), or machine one if necessary out of a small aluminum block. Tomorrow I will try to figure out the threads on the bottom of the antenna and see if there is an OTS solution compatible with this set up. If not, I do have an idea for an adaptor that could work with this or the flag mount as a base. Will talk to @raffimadalian10 at meeting tmr.

m-erez, 2025-01-30:
Antenna model: https://www.amazon.com/Hexa-Boost-Antenna-Outdoor-Omni-Direction/dp/B0953Z1QF3/ref=asc_df_B0953Z1QF3?mcid=33fd022ec2b73b0280edf0065d7f86eb&hvocijid=8279608552387651661-B0953Z1QF3-&hvexpln=73&tag=hyprod-20&linkCode=df0&hvadid=721245378154&hvpos=&hvnetw=g&hvrand=8279608552387651661&hvpone=&hvptwo=&hvqmt=&hvdev=m&hvdvcmdl=&hvlocint=&hvlocphy=9011703&hvtargid=pla-2281435177658&psc=1#aw-udpv3-customer-reviews_feature_div

m-erez, 2025-02-04:
The antenna model is generally supposed to be mounted using u-brackets, according to the linked Amazon page. The cheapest and easiest mount is clamping the antenna to a flag bracket. This one is reviewed to hold well in winds up to 20 mph (+): https://www.amazon.com/Online-Stores-Adjustable-Alum-Bracket/product-reviews/B0014SKUN4/ref=cm_cr_dp_d_show_all_btm?ie=UTF8&reviewerType=all_reviews. 

It's $7.81, so I'd recommend trying this before building a custom attachment to the marine antenna mount linked previously in the thread, which is only a better option if the flag bracket can't handle high wind speeds (reviewers state that it generally can).

The size of the holes on the flag mount are not described, so I can measure them as soon as it arrives. Best case scenario, there is clearance for the standard bolt size. Worst case scenario, I can build a plate with the correct hole sizes, and attach this one to it.

m-erez, 2025-02-17:
This week, I created a fusion model of the assembly, using given dimensions of flagpole bracket on amazon and measured antenna dimensions. I will add this to the NaviGator Full Assembly model/redesign folder soon. Here are some attached screenshots: 

[image]

[image]

[image]

amcaleer271, 2025-02-17:
Looking good so far! Once you get it on the boat and verify everything fits we can move forward with buying the parts.

m-erez, 2025-02-23:
[image]
Added the antenna to the boat, as seen in attached screenshot. I am unsure about the hole pattern in the flagpole mount (no details provided in listing). Likely, either a custom baseplate will need to be attached to it, or holes could be drilled to match the 80/20. Once the part comes in, I will figure out a solution for that. Am also currently working on the model for the second larger antenna, which does not have a mounting system already on it. I'll put together an assembly using McMaster parts before the next meeting.

m-erez, 2025-03-10:
I've been waiting for the flagpole bracket to arrive. I heard it arrived after the mech-general meeting in the latter half of the week, so I'll come in tommorrow evening (if lab is open) to measure the hole pattern. I started the CAD model for the baseplate, but need to add the clearance holes for the holes the bracket currently has. So far, I have three potential ideas for how to create the baseplate:
- print with PETG
- waterjet part (aluminum)
- sheetmetal/drill press part (aluminum)
After I see the mount in person tommorrow, I'll pick one.

m-erez, 2025-03-17:
Finally picked up the flagpole mount last meeting! I tested it with the antennas, and it's fairly sturdy. I will likely replace the wing-nut. 

[image]

This week, I'll finish the CAD for the baseplate, and attach that to the model. I'll get it water-jetted after break.  

I also talked to Lester about the ground version of this antenna. He mentioned that he would like it to be mounted to something that easily sticks in the ground, so I'll research some tent stakes that it could be bolted too.

m-erez, 2025-03-31:
Updated the flag mount model in fusion! I'll put it on the boat soon, someone else was editing the boat so fusion wasn't letting me do while I was modeling. 

For now, here's the sketch the aluminum base adaptor I plan on adding to the mount: 
 [image]
(model will be uploaded soon)

For the other antenna with a larger diameter and no convenient U-bracket system, it might be nice to make a funnel shaped adaptor that lets it set screw in the mount. I drew that above. This piece could be 3d printed or potentially machined from aluminum using the lathe and an aluminum cylinder. 

Lester informed me of a final antenna that is the same one as the RF-kill, but ground-mounted. For this one, I'll research a nice ground stake to clamp it too. Something like this image:

[image]

m-erez, 2025-04-07:
We're on track to finish this week!
Heres the game-plan.

**Today:** Finished modeling the 80/20 adaptor plate on Fusion, it'll have ovular holes instead of circular holes where it meets the 80/20 for more flexibility/tolerance. I'll wait to add the holes to bolt the plate to the mount until the mount gets those holes drilled with the drill press (then I can trace the holes and drill press the final holes on the plate).

**Before mech meeting:** Get plates water-jetted. Order ground screw stakes for the RF Kill antenna on ground (https://www.amazon.com/Orange-Screw-Ultimate-Ground-Anchor/dp/B01D3UI7K8?ref_=ast_sto_dp). The link I posted is supposed to be the best for soft ground/sediment, which seems like the most common environment for the antenna.

**After meeting:** Drill press flagpole mounts and final holes on (1) plate. Test hose clamps with larger diameter antenna. Make sure ground stakes got ordered, and potentially order PVC pipe to create a mount fixture to clamp the antenna onto that's press-fitted into the stake's hole.

Next week, I'll finish up anything that goes wrong this week (for example, alternatives to the hose clamps if those are too weak to support the antenna, and finishing up ground antenna mounting system).

m-erez, 2025-04-17:
Water-jetted and drill pressed out the slots/holes. Might need to pick up a matching material set of slightly longer 10-24s than the ones listed. Will order another flagpole mount and the ground flagpole mount. 

[image]

m-erez, 2025-08-25:
What is the diameter of the cable that needs to come out of the bottom? Or where might I find the cable in order to measure/test the current mount? 

This particular mount actually already has drainage holes in the upper part, but they are pretty small. I can easily drill holes in the highlighted spot too. 

[image]

m-erez, 2025-08-28:
Discussed cable placement with Jack. [image] The plan is to add a hole here for the RF Kill antenna mount, and then zip tie the cable to the base with a bit of slack (which will ensure the cable is at a nice position to travel across 80/20). Then this mount will be done, other than finding bolts/nuts that lock it down nicely. 

Update on the WiFi antenna: I've been told the pvc pipe might be bigger than necessary. I was planning on clamping it to the mount as pictured above and maybe adding a counterweight if the resulting moment pulls it down too much. However, this might not even be necessary if we can get a pvc pipe with a smaller diameter. This is obviously limited by the size of the antenna itself. At some point in the near future, I will venture into storage with someone who has codes so that we can find the actual antenna and see if the pvc casing can have a smaller diameter. If not, the plan will proceed as outlined above.

adamhamdan8, 2025-11-21:
put this on the boat!!

m-erez, 2025-11-21:
Oh yes! I will do that immediately when I'm back from break. They are basically ready to be deployed.