uf-mil/mechanical #26: [Mounting] Camera Mount

amcaleer271, 2024-11-22:
The current camera mount for the front facing camera on NaviGator is 3D printed and currently cracking. 

Requirements:
* The mount should maintain the current camera orientation
* The mount should be able to be attached to 80-20 aluminum extrusion using 1/4-20 hardware.

amcaleer271, 2025-01-23:
Assigned to Ryan Stewart

amcaleer271, 2025-01-23:
Assigned to Bryan

BryanBoc, 2025-02-03:
[image]
Here's what we have for the camera mount assembly right now. We took the tube from the old NaviGator camera and made a clamping mechanism to hold it in place inspired by the ones on Sub 9. We just need to order the tape to wrap around the inside which Ryan looked into. I talked to one of the software girls and she said 15 degrees downward would be good but we can always change that or anything else if needed.

RyanSt3wart, 2025-02-03:
Next steps moving forward are to order the tape and start printing the 3Dp parts out of PLA to test if the tolerances are correct. We will also need to check if we have the correct M2 screws for the sled to mount onto the tube end caps and the 1/4-20s for the mounting.

RyanSt3wart, 2025-02-14:
All printed parts have been printed and tolerances have been fixed on the sled to ensure accessibility and stability. We still need to order the grip tape and check on what screws we need to order once the metric screws are sorted by leads. We also need to talk with leads to discuss how to mount the camera mount on the 80-20 structure for the top of the boat.

BryanBoc, 2025-02-17:
[image]

RyanSt3wart, 2025-02-23:
[image]

Need to find correct M2 screws to attach polycarbonate cover and change mounting holes on bottom to fit with 80-20 structure on top of boat.

BryanBoc, 2025-02-23:
^^. Also need to change tolerance of holes on reprint.

RyanSt3wart, 2025-03-01:
Changed the mounting holes on the legs to line up with the 80-20 structure. Need to combine in CAD before 3D-Printing the final designs and may need to still change the tolerances of the holes.

BryanBoc, 2025-03-03:
Finishing up CAD for final design which will fit onto the 80-20. Will check with one of the leads before printing. 

[image]

BryanBoc, 2025-03-10:
Gonna try to get this fully printed this week

RyanSt3wart, 2025-03-10:
Need to combine the new bottom parts of the legs in cad then do final print.

RyanSt3wart, 2025-03-14:
Finished changing holes and configuring new legs in CAD. We are ready for the final print. We are waiting on PETG filament to arrive from Bambu

RyanSt3wart, 2025-03-29:
Setup printer with new PETG from Bambu. Currently printing final updated prints.

BryanBoc, 2025-03-31:
^ Gonna try to get it fully printed and assembled early next week.

BryanBoc, 2025-04-07:
Still getting this printed... had some problems with the printer but I think we figured it out.

RyanSt3wart, 2025-04-07:
^ This + Added additional supports to the slice

BryanBoc, 2025-04-21:
fully printed, needs to be assembled and put onto NavigGator. Last time I was in Cimar the 80-20 was not the same as the fusion but I believe they're still working on it so I'll check it out this week

RyanSt3wart, 2025-04-21:
Still need to drill the holes out to .265 for the 2 top camera mount clamps.