uf-mil/mechanical #28: [Mounting] Hydrophone Mount

amcaleer271, 2024-11-22:
The current mount for the hydrophones is not functional and is hard to use

Requirements:
* Rigidly mount the hydrophones so they can take measurements under the surface of the water, they don't need to be deep, just fully submerged
* The hydrophones must be mounted in a location where they are not directly in the thrusters line of action.
* The hydrophones should be able to be removed from the mount without any tools
* The mount should be able to fold up or retract and be stowed securely during transit.

DiegoDom23, 2025-02-03:
The hydrophone mount used in the last competition was essentially a grip mechanism attached to an aluminum bar, which caused several stability issues. For the new boat, we want to ensure the system is more flexible, making it easier to transport. Additionally, considering the testing conditions at Lake Wauburg and the next RobotX competition being in 2026, we want to ensure that whatever materials we use can withstand corrosion and last long without needing replacement.

I have a couple of ideas on how to implement a system that meets the requirements, and I’ll be doing some CAD modeling as well as talking to team members to figure out which mechanism will work best for our new boat.

The first idea is to use a hinged bracket that folds down over the side of the boat and locks into place. This is similar to what we had last semester, however, the approach will be slightly different because the rod last year was too stiff and caused several issues. The hydrophone would be mounted with a quick-release mechanism that provides enough security to hold it in place while the boat is operating, but also ensures the removal process is simple. This could be done with a magnetic clasp that tightly secures the hydrophone.

The second idea is to use a telescoping arm that can extend long enough to clip the hydrophone at the end of the arm but can also be easily retracted to a minimal size for safe transport. Additionally, if the telescoping arm were placed on a hinge, it could fold up so that, during transport, the arm would lay horizontally (pointing parallel to the water). When in use, it could then fold down (pointing perpendicular to the water) and be extended to submerge the hydrophone underwater. This idea looks promising on paper, but executing it will be tricky. However, since many objects use extending arms, we can take inspiration from those designs.

The third idea, and probably the riskiest one, would be to use a retractable reel system, similar to a fishing reel, to hold the hydrophone mount. With this system, the hydrophone could be easily lowered and retracted, while the hydrophone mount would be clipped onto the cable in a way that makes removal simple. The reel would also be easy to retract, ensuring that it doesn’t interfere during transit. There are obvious limitations with this idea, including the strength of the reel, the hydrophone dangling at the end of the reel while the boat is operating, and how to ensure the mount is properly secured onto the reel.

A big part of this is also working with the people in charge of building the thruster mounts, as their locations will decide where the hydrophone mount should be placed. Ideally, it would be towards the center of the boat to ensure it has a great gripping point.

I have talked to Cade Sachs, another member of the mechanical team, and he has agreed to help me out throughout the design process.

DiegoDom23, 2025-02-17:
So far, for the hydrophone mount we have sketched a couple designs, with the best one being the one shown below. There will be an arm, that will just be a simple bar or tube that will be held onto the boat with a pivot joint with locking handles. The pivot joint will be able to bring the arm up and down. When we have to transport the boat, the arm will go up, causing it to be held horizontally in a safe position. Once we are in the water, by using the handle one can simply bring the arm down to make the bar holding the hydrophone tube go into the water. The hydrophone tube, alongside the hydrophone, will be held in place with simple worm drive clamps as of now. We still have to consider how easily removable this will be, as we know one of the design requirements was to be able to remove it without tools, but as of now this is what we came up with. The CAD model currently does not have the proper dimensions in place, so the next part of the process is ensuring everything will fit. We are also banking on the boat having a bar that goes across the frame to hold the thrusters, like we did in the old version of NaviGator. Without that bar, the placement of this structure would be hard, and we will probably need a brand now design. 

[image]

[image]

[image]

csachs32, 2025-02-17:
For clamping the mount to the beam on the boat, we are still unsure of how we'll mount it because we are waiting for the thruster team to decide what beam they want to use. Also, for the extended bar, we are thinking about an extending arm stand that we have found online. It is a simple metallic round bar that can extend to different lengths. We don't have to use this exact bar, as there are probably other better other options but we feel this idea is the best. We can make another separate mount that mounts the hydrophones to the extending arm as well. It could be something like this (either the arm or the stand could work): https://www.amazon.com/AmazonBasics-Tripod-Boom-Microphone-Stand/dp/B019NY2PKG/ref=sr_1_1_ffob_sspa?crid=1R467NN3C11IT&dib=eyJ2IjoiMSJ9.MokwjC4C4HCXvVYzzdd6JAR_9de--nEcql21WWTLH3pQ1S6JN08JRUKyBhQl3hnIe4oTNW3XMMiYEe84yIY0ArAY1_fg__OFWiEuodom6WquEMZ5ZeTBddnRplioqrH2CYvGjAxPKY2PIFwiHtI1NneCyI8cHtA3KDhsGDgl2gJ45w6SKJKT62OleRNMPYO1HQoVSUXrgNy7SnC7l3y4HjtFpVJuWxeXKO7DNIhXfIV53IdpdhstayPO2jL-TXcbOYmZeUtlUANd6d7Ls3J_HvGc67EXcczlaKSRWKnYV5Y.K_DhqOFan9TXqV_XmwP__MB6bbA6-ojnaUYFTVArxIw&dib_tag=se&keywords=mic+stand&qid=1739767475&sprefix=mic+stand+%2Caps%2C102&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

amcaleer271, 2025-02-17:
My concern if I'm understanding the design so far is that there is only a small pin that is going to be taking the full force generated from the hydrophones moving through the water. The boat can move pretty fast, so a large moment can be generated about the pin.

DiegoDom23, 2025-02-24:
The moment generated by the water was definitely a factor we considering in the first design considerations, so I am planning on making sure there is enough clamping force to ensure the extending bar will not fall off. I have not generated a full CAD model of new additions to the idea yet, but I will try to have one for the meeting this week. A few design ideas could be implementing a secondary clamping mechanism like a split collar with a bolt-tightened clamp that will go around the pivot to add friction to reduce the total force felt by the bar. The material of the pivot pin is also going to play a big role on how much force it can withstand, so ensuring we choose the correct material will go a long way. The bar holding the hydrophone should not be too long either, and the material it is made out of will also play a large role on if it will fail or not.

amcaleer271, 2025-04-10:
Both Cade and Diego ended up leaving the lab mid-semester. Someone else can pick up this task if interested.

amcaleer271, 2025-08-25:
The hydrophones need to be completely rigid while in the water, a similar mechanism to the thruster mounts would likely work well

AlexunderscoreM, 2025-09-05:
I wanna do this

closed 2026-01-03 by amcaleer271