uf-mil/mechanical #29: [Structural] CPU and GPU Cooling

amcaleer271, 2024-11-22:
The CPU and GPU in the computer can get extremely hot, and this can both throttle performance and be dangerous. The existing cooling solution will need to be modified to fit the new electronics box. 

Requirements:
* Find a way to use the existing water cooling loop to maintain low temperatures of the CPU and GPU
* The solution must either be attached to the CPU/GPU Box or easily detachable so the CPU/GPU box can still be removed and worked on in lab

Adam Hamdan worked on this previously, see the archived issue or talk to him if you need help

adamhamdan8, 2025-01-25:
Its pretty simple and already set up, just needs possibly new tubing and a some new fittings to meet the new space and location requirements

amcaleer271, 2025-01-28:
For the high power box and the MIL board box, @nonagaga is working on finding PC case fans to cool the boxes, as they shouldn't generate a large amount of heat (and were uncooled on previous NaviGator). 

The computer box is going to be cooled with water cooling as it was on previous NaviGator. The radiator will be attached to the pelican box so the whole unit can be removed easily (for work in lab or debugging).

space-cat-explorer, 2025-01-29:
Talked with @nonagaga about making sure the external fans do not cause issues with possible water leakage into the boxes. Solution would be to implement some sort of covering above and below (or all around) to protect from splashing will maintaining airflow. Could be a 3D printed part.

adamhamdan8, 2025-02-27:
[image]
Here is the microatx diagram

nonagaga, 2025-03-03:
24V case fans, standard GPU air cooling, and an interior water cooling AIO loop were ordered for the first iteration of computer box cooling on NaviGator. The footprint of the box was also increased to accommodate for the extra space the fans and radiator took.

amcaleer271, 2025-08-25:
We need to get this done pretty soon so we can start testing the boxes

We could use an AIO to make this a lot easier! @nonagaga thoughts?

nonagaga, 2025-10-20:
Started CAD-ing a design for the radiator mount. A 3d printed guide plate will be fastened to the back of the radiator, which will slot into a mount on the box itself. This allows us to easily remove the radiator, so we can remove the computer without uncoupling the radiator.

adamhamdan8, 2025-10-27:
just for documentation we made a separate issue for the GPU air cooler reinstallation, which is #122 , and im working on that

nonagaga, 2025-11-10:
[image]
[image]

Got the adapted piece from Ricky, radiator cooling mount for NaviGator is complete! Pls review and close issue if good.