uf-mil/mechanical #31: [Structural] New thruster mounts

amcaleer271, 2024-11-22:
The thrusters on NaviGator are currently all positioned at 45 degrees. This allows omnidirectional movement, but significantly slows us down when trying to move forward. Furthermore, the current thruster mounts are old, have unnecessary components, and at times damage the thrusters.

Requirements:
* Find or create a mounting solution to position the thrusters are 30 degrees from forward (the general direction will remain the same as the current thrusters)
* The mounts should make it easy to raise and lower the thrusters both on land and in the water, consider raising the thrusters linearly instead of at an angle.
* In addition to raising the thrusters manually, a nice feature would be to raise and lower the thrusters remotely.

amcaleer271, 2024-11-23:
Consider using a linear actuator like [this](https://www.amazon.com/dp/B0D2NHYBNS?ref=emc_p_m_5_i_atc&th=1) to raise and lower the thruster

amcaleer271, 2025-01-23:
Blake, Harin, Diego, Lucas

bcobo04, 2025-01-24:
Regarding the ability to raise and lower the thrusters, a very simple solution to implement with the new mounts would be something similar to a gym bench's spring loaded pin that allows you to pull out the pin partially without fully removing it in order to adjust thruster height vertically. Examples of the pins seen below. Once near the position, letting go of the pin and slightly moving the height would cause the pin to drop into place being pushed in by the spring. Only two preset positions would need to be made for the pin to slide into giving an option for an "up" and a "down" position. The implementation of the pin regarding its location will likely depend on the mount itself. There is also a potential problem with using the pin through the PVC as there are wires running through that could possibly get pierced.

[image]

[image]

bcobo04, 2025-01-24:
Oh ok from the meeting the other day you and Sophie made it seem like automating it was overkill and unnecessary. I talked to Sophie today and she cleared it up saying Schwartz was the one who wanted it automated so we'll definitely look into it when designing the new mount parts. But also it seems like the priority is to get the thrusters mounted ASAP just so we can actually test the boat.

bcobo04, 2025-02-02:
The attached image shows the concept that was discussed at the last meeting to allow for the use of spring loaded pins to change the height without interfering with the wires in the pvc pipe. Hopefully it makes sense even without the design for the actual mount being done to connect it to the boat. At the bottom I talk about problems with the discussed design and offer a possible solution. Any thoughts?

[image]

lucas-reilly, 2025-02-02:
**Locking Mechanism**
Gave thought to the locking mechanism; ideally we would use a standard pin + spring part as pictured by @bcobo04 regarding the concept. There is need a locking mechanism for holding the thrusters given the system drawn by @bcobo04 so I decided to sketch it. We still need to figure out the current mounting situation for the boat so that we can integrate the locking part easily. The blue Saturn shape in the middle is the thruster attachment pictured in the sketch above.

[image]

lucas-reilly, 2025-02-06:
Latches for sale:
[McMaster](https://www.mcmaster.com/products/spring-plunger-latches/)
Limited selection in length, higher reliability
[Amazon](https://www.amazon.ca/s?k=spring+plunger+latch&crid=2WL4ZKAA4ZP4J&sprefix=spring+plunger+latch%2Caps%2C119&ref=nb_sb_noss_2)
More versatile in sizing and style, weaker and less reliable.

bcobo04, 2025-02-09:
I made the design discussed above in CAD. Obviously going to be changed as this chunk of metal is not feasible or practical.
[image]

bcobo04, 2025-02-09:
Additionally, I went to the lab to check the new metal parts of the pontoons to see what the mount will be built on. I have dimensions that I will be using in the mount design and pictures of the back for reference.

[image]
[image]
[image]

Harin-0253, 2025-02-10:
I had an idea for a way to automate the raising and lowering of the thrusters that I think can work well. Using the brackets Blake designed above, a winch like the one pictured can be threaded through the top of the thruster and the winch, with the bracket securing it, can effectively raise and lower the thruster. The winch can probably be attached to the top area of the boat.

[image]

Harin-0253, 2025-02-17:
Here's my design for a thruster cap that will attach the end of the thruster to the carabiner at the end of the winch. A pin will go through the hole at the bottom securing the thruster and the carabiner will latch onto the loop.

[image]

amcaleer271, 2025-02-17:
Is there another part that will keep the thruster's motion constrained to up and down.

Harin-0253, 2025-02-17:
I was planning on using the bracket Blake made earlier.

bcobo04, 2025-02-24:
This week I've been working on design concepts for the actual mount part to attach the boat to the part I previously made. Nothing to show yet but I should have at least a few sketches of different ideas or possibly a model by next meeting.

bcobo04, 2025-03-02:
After going over design choices and possible solutions with the leads. It seems there are better ways to go about mounting the thrusters. After doing some research, I was able to find an easily integrated solution off the shelf that can be automated with a small easy upgrade. For the actual thruster clamp, the leads and I agreed that the friction clamps used previously worked fine and would not be a problem so we don't have to use the pin-based clamp designed previously. With some slight modification to the old friction clamps, they will attach to the track system linked and pictured below. (We will have to decide how much travel we actually need for the thrusters so we know which size to buy) This part allows for the raising and lowering of the thrusters manually through the crank on the top, but can also be fitted with a motor to spin the mechanism automatically. Then to connect the track part to the boat we will need a trapezoid shaped block much like the one from the previous NaviGator iteration. I will begin work on modeling the new friction clamps to secure onto the track this week.
https://www.ebay.com/itm/315594352157?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=sckwsrrlqze&sssrc=2047675&ssuid=&var=614037852620&widget_ver=artemis&media=COPY

[image]

Harin-0253, 2025-03-03:
If we’re going to pivot away from a winch based system I can begin working on a way to automate the crank for the system Blake is talking about.

lucas-reilly, 2025-03-09:
I have been working on an attachment point for the rail that Blake is talking about. I decided to angle the rail with respect to the boat because that is the way the thruster is going to be pointed. If it would be pointed any other way there would be excessive shear force. I am hoping to CNC this part out of aluminum for strength and will fix the rail upwards with bolts to the side. I might also add flanges for stability depending on further conversations with the team. I need to discuss if the current rail mounting system will be strong enough to hold the motor as it is only a single block on a rail. There might need to be an auxiliary mounting point. In the future, I am looking to clean up the design and determine the size of the hardware that I will be using.

[image]

bcobo04, 2025-03-10:
Remade old clamp model in fusion as it was previously made in solidworks preventing me from editing it to update it to connect to the new rail system. New dimensional adjustments and outcroppings will allow it to directly attach to the rail part I recreated in fusion previously as seen below.

Remade Friction Mount:
[image]

Rail Block:
[image]

Harin-0253, 2025-03-10:
found a motor that is apparently directly compatible with the crank. Also upon closer inspection the actual crank part of the hand wheel seems to be able to detach, which would make integrating the motor really easy. Just need to see what the detached hand wheel looks like.

Motor: https://www.amazon.com/UIROBOT-Stepper-Integrated-Controller-24-36VDC/dp/B0D59STYHP?th=1

Harin-0253, 2025-03-15:
I designed a bracket/mount for the servo I linked. The mount should work on 80/20 rods, and keep the motor secure. The bracket has two parts, a cap and sleeve. The cap fits onto the front of the servo and screws into the sleeve which is on the other side of the holes. The sleeve has three holes allowing for 1/4 20 bolts to fit in order to secure the motor to a rod of 80/20.

[image]

[image]

bcobo04, 2025-03-16:
This week I have begun modification of the original clamps to mount them on the rail block. Fusion is being difficult and the difference in dimensions between the parts being inches vs mm is not helping. Made siginficant progress though and will likely be done with the model by next meeting. After a working model is developed, I recommend purchasing the rail system and printing at least 1 half of the friction clamp to ensure it will fit on the block with no problems before machining the new clamps.

[image]

bcobo04, 2025-03-30:
Clamp CAD is fully done and is ready to be looked over. I will print a 3D print a prototype so that it can be tested. Probably need to mess with some dimensions because some things seem bigger in CAD than they actually are making for some really small features that would break in reality. I will also be putting in the order for the rail system so that we can begin assembling the new mount system. I'm thinking the 900 mm of travel will be enough unless @amcaleer271 @Harin-0253 or @space-cat-explorer have a differing opinion. Once put together and tested, we will decide if this is viable for the rest of the thrusters and begin work on that. 
Link to 900 mm rail again: https://a.co/d/4IWbaNV

[image]

lucas-reilly, 2025-03-31:
Made slight changes to dimensions on the rail attachment point to the boat so that the rail doesn't hit the mounting point, added flanges to the sides for better support, and created a flipped version for left and right sides of the boat. The primary angle is for the mount point to the boat, the secondary angle is to minimize shear stress from the motor onto the rail mount. Need to confirm specific hardware sizes in the shop for bolts, and need to confirm dimensions. After this, I'll 3d print a prototype and test it when the rail comes in. Once the 1st prototype is tested, it will be CNCd out of aluminum for greater strength. As this is a critical attachment point on the boat, it needs to not fail. I will be in the shop early this week to talk to team members about fabrication (flanges vs no flanges, welding, hole cutting).

[image]

amcaleer271, 2025-04-03:
Because the parts and prints for this are big and/or expensive, please add all the parts to the boat assembly so we can verify everything fits and review it.

bcobo04, 2025-04-03:
@lucas-reilly Adam says that the part you designed, while it would work, can't really be CNC'd and also getting access to the CNC for that is going to be difficult. Additionally, if we did want to machine it by hand, the angle is going to be difficult to do. So we are gonna table that part for now and were pivoting this project a bit.

bcobo04, 2025-04-03:
IMPORTANT: After going over it with the leads, we are going to pivot the thruster mount project. Because the back thrusters are going to be different as we are upgrading to the military ones, this system is going to be adapted for the front thrusters. @lucas-reilly start thinking and maybe CAD some ways to mount the actuator to the crossbar in the front of the boat. (The part can be found in the old boat CAD on fusion). I am ordering the actuator and we may have to disassemble and make our own block that translates up and down if the one on it is too small. (Which its looking like it might be and also we don't know the material). The front thrusters have a lot less variables that can go wrong so that is the goal for the system. As soon as the actuator comes in, we can decide what we'll do with the block and I'll adjust the friction mounts to fit the new mount.

lucas-reilly, 2025-04-07:
[image]
Prototype 1 of new thruster mount to front crossbar

Need to talk to boat leads as to whether to incorporate angle into this design or simply turn the motor instead. This system will lock the actuator in place. The white in the design is the existing material. This should hold all weight if cncd, will 3d print a prototype for fitment. Looking to confirm hole sizes in the future and talk with @bcobo04 and @amcaleer271 as I was out last week.

bcobo04, 2025-04-13:
This week was slow as we are waiting for the delivery of the linear actuator. We can't really progress much without the actual hardware. We did go to check on the old crossbar for the front thrusters, but it was at the bottom of the boat box so we need to get a group to move the stuff on top so we can get measurements. If the old crossbar has too many holes in places that hinder our ability to secure the new mounts, we could just order a new crossbar that has a length of around 8 feet.

lucas-reilly, 2025-04-14:
Also a slow week for me as I am waiting for the actuator to begin manufacturing the design for the front thruster mount. I don't believe the holes in the crossbar mount should be a problem, but it should be reexamined anyways to see if the current design gets in the way of transport and to confirm specifications.

Harin-0253, 2025-04-14:
Slow week as well waiting for the actuator, but once it arrives I can verify the motor pairing for it and finalize my mount designs and the motor order.

bcobo04, 2025-04-20:
Actuator has arrived. There are a couple of design routes to choose from. I will outline a couple of them in a later message in case the current "frontrunner" falls through or proves itself difficult to implement for the summer team. This idea involves fixing 2 L-brackets to the block using the hex screws that are already there, but getting longer ones to account for the addition of the L bracket thickness. These L-brackets will provide the base for which the back half of the friction clamp will be attached. I have attempted to draw this design below, but if anything is unclear please feel free to dm me (Blake Cobo) on discord over the summer and I will respond to clear up any questions you may have. 

[image]

bcobo04, 2025-04-20:
**FOR SUMMER PPL:** 
The linear actuator rail system was last left under the tables by the window in the front of the lab. The block that moves with it has been taken off so we can design around it. The block was last placed on top of the tables by the sub in a plastic bag so the ball bearings inside are not poured out. Please handle with care so the ball bearings are not lost. Pictures are included below to help locate them. Please don't hesitate to dm me (Blake Cobo) over the summer with any questions about the project I'd be happy to help.

[image]
[image]