uf-mil/mechanical #32: [Sensing] Thruster Feedback

amcaleer271, 2024-11-22:
The thrusters currently operate in an open-loop manner where a command is sent to the thrusters to spin at a certain RPM, but we get no feedback on how fast the thrusters are actually spinning. This can make debugging issues difficult and time consuming.

Requirements: 
* Find or create a method to get RPM feedback from the thrusters
* Collaborate with software and electrical team to ensure the solution will work and is able to be implemented

nonagaga, 2025-01-23:
Here's a repository detailing the torqueedo TQ bus protocol to read/write motor control values.

https://github.com/ArduPilot/ardupilot/blob/4e12b4ee0e4618a8093bfa4705b3ce2bffba52fc/libraries/AP_Torqeedo/AP_Torqeedo_TQBus.h

Here's the documentation from that repository on setting up the circuitry required to interface with the actual can bus.
https://ardupilot.org/rover/docs/common-torqeedo.html

The project (Ardupilot) says only a single thruster can be used with their implementation. We'll likely have to do some wrangling to get it to work our way.

amcaleer271, 2025-08-25:
Not a priority right now!

closed 2025-08-25 by amcaleer271