uf-mil/mechanical #33: [Structural] Status Tower

amcaleer271, 2024-11-22:
The current status/light tower on the boat can be difficult to see at times, and is not very securely mounted .

Requirements:
* Collaborate with electrical team to find or create a solution for making a more visible status indicator solution
* If the solution is mounted to the boat, ensure it can be mounted to 80-20 aluminum extrusion using 1/4-20 hardware.

amcaleer271, 2025-01-23:
Assigned to lana

l-gerstenberger, 2025-01-27:
Began research on the LED strips as a possibility to replace the status tower. However, most LED strips being sold currently do not reach the amount of lumen needed to be visible during the day. These LED strips seem to be limited to around 1000 lumen, but the amount needed to be visible during the day is around 3000-5000. Instead, will begin research on more powerful status towers next week.

l-gerstenberger, 2025-01-30:
Found Embry Riddle’s component list and they use this led panel. Uses 12V https://www.ledlightinghut.com/8x32-neopixel-ws2812b-led-matrix.html
Data sheet: https://www.ledlightinghut.com/WS2812B-Datasheet.pdf
Also looked through the other top teams’ component lists and found that the FAU team uses this one.
https://www.bannerengineering.com/us/en/products/part.19278.html

l-gerstenberger, 2025-02-03:
Decided on the LED panel to replace the current status tower after looking through RobotX teams and liking Embry Riddle's lighting panels. Just need confirmation that it would work well with the battery and that the draw is not too much.

l-gerstenberger, 2025-02-10:
Determined a mount design that should work to hold the LED panel in place. Just need to finish drafting it and put it in the assembly.

l-gerstenberger, 2025-02-17:
Inserted the design into the redesign assembly, and the mount is compatible with the 80x20 aluminum. The design may be too thick, however, the website where I found the LED panel did not have a thickness for the panel listed, so I estimated what that measurement was. The design is also compatible with the 18-8 stainless steel hex drive screw.

amcaleer271, 2025-02-17:
They look good, the place you'd put them in the assembly would be a bit risky as that's where the electronics boxes will be, so we'd have to be careful not to hit the LEDs when taking the boxes in or out. It may be better to place them higher to avoid that, and also to make them more visible.

l-gerstenberger, 2025-02-24:
Submitted the order form for the LED panels and am shifting now to working on the SubjuGator prop.

l-gerstenberger, 2025-03-03:
Spoke with members of the electrical team working on the status tower project. They said that we do need to have a status tower, and that the LED panels are not allowed on their own. We found an alternate status tower that will produce more lumens, so I will continue communicating with them and create a design for a mount.

l-gerstenberger, 2025-03-09:
Finalized the decision for LED panels with the electrical team. Once the panel arrives, we will test that it works with the electronic configuration and then order more as needed. If it does not work, I will research some alternatives. Also, will finalize the mount design with the electrical team and ensure that it will be sufficient for the wiring of the LED panel.

l-gerstenberger, 2025-03-16:
Waiting on the LED panel to arrive but once it does, I will take it to the electrical team and learn how they intend to test it. From there, I will order the remaining three and can 3D print or machine the mount to go with it.

l-gerstenberger, 2025-04-06:
Heard that the LED Panel has arrive and updated the electrical team on it. I asked them to keep me updated on testing and to keep me involved in the testing process, so I can learn from it. Will check up on the panel in person on Tuesday.

l-gerstenberger, 2025-04-21:
The LED panel is working and the electrical team and I will meet up to wrap up this issue

l-gerstenberger, 2025-04-22:
Final Report: The status tower used for the last competition was reported to be too dim so that the team members on the shore could not differentiate the colors being displayed. To solve this, I first researched the different options for tricolor vertical status towers and found that we were already using one of the brightest options on the market. Furthermore, the brighter options would not be too different from what was already being used, and therefore would be a waste of money. I instead found that successful teams like Embry Riddle were using LED panels, which illuminate a greater surface area and therefore can be visible at further distances. I next designed a lightweight, simple but reliable mount for this LED panel. The mount is made up of a front plate and a back plate, with the front plate cut out so that the LED is visible. The two plates are simply fastened together using a screw that can be mounted to the 80/20 extrusion that is used for the chassis. At this point in the semester, the electrical team has tested this LED panel and confirmed that it will work with the electronics currently on the boat. The only remaining task is to 3D print the mount and assemble onto the boat.

amcaleer271, 2025-08-25:
If this is ready to go we can finish printing the mount and then get it on the boat!

l-gerstenberger, 2025-09-29:
Checked on the printed mount and talked with Adam and Erik about changes to the design. Decided on altering the way the mount seals to allow me to use an epoxy to seal in an acrylic plate. Will make these changes and reprint.

adamhamdan8, 2025-10-27:
okay heres my idea rn but pretty rough. sorry for not sketching this sooner, and sorry for this being in ms paint im too lazy to get my ipad but i gotta do this now before i forget again. this is a cross section but this is the idea. but also yeah you would have to break the 3d printed part or something to get it out. this is just the idea tho

[image]

this is lowkey a horrible sketch so just let me know if I can describe or help you model it or whatever I really don't mind.

l-gerstenberger, 2025-10-31:
This was very helpful, and I think I got the gist. I modeled it up, so let me know if there's anything I should change or anything I got wrong. I made the cutout for the acrylic 0.093" based on a sheet of acrylic I found that is sold at Lowe's. Also the only thing that I did differently from the drawing is I made the cables run out the side, but if you think it would be better if they come out the back, I can make that change. Right now there is a indentation in the back of the mount and then a cutout on the side to allow them to attach to whatever.

[image]

[image]

[image]

l-gerstenberger, 2025-11-10:
I changed the orientation of the indentations for the cables as suggested by Adam. I also adjusted the bottom plate to not need the top plate, as it will be sealed by the epoxy.