uf-mil/mechanical #36: [Organizational] Organize the screw boxes

amcaleer271, 2024-11-22:
Someone has to do it

ryang412, 2026-09-11:
HOLD ON TO YOUR HATS ITS HAPPENING

ryang412, 2026-09-18:
1/4-20 and 1/4-28 bolts and nuts organized by size and type

[image]

ryang412, 2026-09-29:
screw boxes printed and moving towards metric organization and #6-#10 imperial

ryang412, 2026-10-05:
6-10 imperial boxes and m3 boxes completed!

Thank you Ali 

[image]

ryang412, 2026-10-19:
This [organizer](https://thegarageoutfitter.com/products/montezuma-4-drawer-garage-base-cabinet-30-5-x-24?currency=USD&country=US&variant=40444647342212&utm_source=google&utm_medium=cpc&utm_campaign=Google%20Shopping&stkn=e3ff745be00a&srsltid=AU7gw4ULELLzeBWzT0ha3qL5zl2MOs_N4tuDBPjmgj2lFlMHDwu8rVZhJDs) might be just crazy enough to work. Proceed with caution

ryang412, 2026-10-20:
The hardware is organized. Please don’t put stuff back random, I made empty miscellaneous boxes specifically for when you don’t know where to put something. If you are rough with the yellow boxes, the hardware goes everywhere, especially metric box, so handle with care 🙂 When in doubt, consult fasters map. 😎💯🔥 Thank you Lorenzo, Ali, Isa, joli, and friends!

[image]