uf-mil/mechanical #37: [Organizational] Tool inventory/refresh

amcaleer271, 2024-11-22:
We use a lot of tools in MIL, however they have a tendency to disappear. To keep the lab productive, it's important to have all the tools we need, and know where they are.

Requirements:
* Take a comprehensive inventory of all the tools we have in MIL
* See what is missing, and ensure we have all the tools we need (and spares)
* Take note of the sizes of tools we use most frequently, and make sure we have plenty of those

amcaleer271, 2025-01-21:
Some tools we are definitely missing:

- Complete socket wrench set
- Variety of screwdrivers
- tape measures (need multiple)
- Files

space-cat-explorer, 2025-01-29:
Going to take inventory of mech parts on Saturday and get a list of anything we need. Will also be organizing mech work station and any drawers with mech materials