uf-mil/mechanical #38: [Lab maintenance] Fix CNC

amcaleer271, 2024-11-22:
We have a decently large 2 axis CNC in the lab, and while it sometimes works, it is not fully operational.

Requirements:
* Learn how the CNC works
* Make any purchases or repairs necessary to make the CNC fully operational
* Learn how to operate the CNC and document the process so others can use the CNC.

amcaleer271, 2025-01-23:
Raffi is also working on this

RyanHoburg, 2025-01-23:
https://www.omiocnc.com/products/x8-2200l-usb.html

RyanHoburg, 2025-01-23:
Raffi Maddie and I tested the pump and lines, unfortunately the blockage is somewhere in the spindle housing itself

dylan-duenas, 2025-01-23:
The main solutions I've seen to clear the blockage in the spindle housing are running distilled water (or bleach if we're desperate) and clearing it with compressed air. Either way, it seems like we have to run whatever we try both ways before the blockage is (hopefully) cleared.

dylan-duenas, 2025-01-27:
So today I was able to clear the blockage just using the compressor in the lab. Ran the coolant through and it was able to flow smoothly both ways. Now the only thing left to tackle is the mist coolant. I was not able to find the 5/16th (or technically 8mm) hose for the air fitting, nor the coolant hose and inlet for the coolant, so we have to buy these parts. I sent a purchase order 5/16th 20ft long polyurethane hose for the air, and a whole mist coolant setup (couldn't find the proper hose with the brass inlet fitting online, and it's just as cheap to buy the whole system anyway). The links are listed below, so we just have to wait for those to come in and we should be good to go.

[](https://a.co/d/eHFykhs)

[](https://www.amazon.com/gp/product/B071DXGGP4)

adamhamdan8, 2025-01-30:
I ordered some things that @dylan-duenas asked me to buy, I will let you guys know when the Amazon order arrives. I also believe @nonagaga is going to use it to machine the bus bars for the power merge board, meaning the CNC is at least somewhat operational. Also @dylan-duenas try to link the parts in the github issue!

Finally we need to figure out the issue with the pump, it may just be that we need more coolant in the bucket but we want a higher flow of coolant.

dylan-duenas, 2025-01-30:
Tried to run the CNC to mill out the copper bus bars. The first try was ruined because it had to be paused midway. The second try had a lot of vibration caused by the sacrificial board, which superheated the copper so I turned it off for safety (it was starting to  burn the MDF). We should look into drilling out holes in the sacrificial board to leave it clamped down onto the bottom with bolts and captive nuts.

[image]

amcaleer271, 2025-01-31:
Good work on getting it running! While you're working on this, make sure to document the process of running the CNC.

nonagaga, 2025-01-31:
> Tried to run the CNC to mill out the copper bus bars. The first try was ruined because it had to be paused midway. The second try had a lot of vibration caused by the sacrificial board, which superheated the copper so I turned it off for safety (it was starting to burn the MDF). We should look into drilling out holes in the sacrificial board to leave it clamped down onto the bottom with bolts and captive nuts (like the image attached).
> 
> [image]

I worked with Dylan on getting Sub9 bus bars CNC'ed, and was unable to resume machining once paused. Since the part failed anyways, I'd like to try to save it by estimating the zero position by using the cut contours and the Fusion G-Code. I'm not sure how likely this is to work, so we may have to start from scratch with the new copper @amcaleer271 just ordered.

dylan-duenas, 2025-02-16:
Working on a handbook to operate the CNC. Starts with getting the setup right on fusion, then goes into toolpathing, post-processing (and the associated naming convention), and finally, operating the CNC.

amcaleer271, 2025-04-10:
How is this going? If now we're working only on documentation, we can close this task and open a new task for documentation