uf-mil/mechanical #39: [Organizational] Inventory Spreadsheet

amcaleer271, 2024-11-22:
We have a lot of hardware in MIL, between tools, fasteners, stock material, and other components. It can be difficult to find parts needed, or sometimes not even know if we have a part. To rectify this, we want a large spreadsheet to generally track the hardware we have in MIL. The sheet does not need to track the specific quantity of each component, but should give us an indication of if we have things in the lab or if we need to buy more.

Requirements:
* Create a spreadsheet (or similar solution) to track the hardware in MIL
* Each item should have the specific name of the item, some general category (fastener, stock, etc.), and the location in the lab
* Fill out the spreadsheet with what is currently available in the lab.

dylan-duenas, 2025-01-28:
Used a parts tracking sheet developed by FRC team 6328 Mechanical Advantage and modified it for use in the lab. Generates a part number for each part based on its location and type, also has entries for purchases, description, material, manufacturing, quantity, and status. See link here: [](https://docs.google.com/spreadsheets/d/1edr2M4JpRf445QYh4DPXyDiPAV5vay3gAtJp_E842Lo/edit?usp=sharing)

space-cat-explorer, 2025-01-29:
Some of the features of the spreadsheet Dylan provided didn't work, and most of the information is not super necessary. I made a google sheets with just categories that are important, and will start filling it out on Saturday. I will not be doing it for electrical and software items, just for mechanical. I will also be organizing the workbench and any drawers/cabinets with mechanical products in them. 

Here is the link to the google spreadsheet: https://docs.google.com/spreadsheets/d/16-KZdbdkgPLAdMlMyCpAt7bKNaXPOyRIgghhcRyKflA/edit?usp=sharing

Once this is filled out we will know if there is anything else we need to purchase for the semester. Not sure if people are going to actually use the spreadsheet in the future, but it will be a good way to start organizing.

adamhamdan8, 2025-01-30:
> Some of the features of the spreadsheet Dylan provided didn't work, and most of the information is not super necessary. I made a google sheets with just categories that are important, and will start [...]

Make sure to bring this up to electrical and software teams today during our meeting to try and get the whole team on board with organizing our shiz

adamhamdan8, 2025-10-27:
im gonna close this, I think a software member is working on a SQL solution to inventory for the entire lab!

closed 2025-10-27 by adamhamdan8