uf-mil/mechanical #45: Fix Rover

amcaleer271, 2025-01-16:
This can be a task for multiple people

adamhamdan8, 2025-01-23:
Jeremiah and Gabe Hoberman are going to be working on this this sem!

cbrxyz, 2025-01-23:
@GabeHoberman hello! I'm going to be working on the software side of the rover! If you need anything, feel free to let me know!

amcaleer271, 2025-01-27:
I've created a project in the MIL mechanical fusion team to make it easier to collaborate. Jeremiah is working on converting the existing Solidworks CAD to Fusion.

adamhamdan8, 2025-01-30:
I'm going to go ahead and add myself as a point of contact with leadership and Cameron, if you guys need any direction or need any help on any solutions, use me kind of a consultant for the project

SaintFStudios, 2025-02-02:
Week of 1/27/25 Update:

- Done a general overview of the rover, including previous documentation, videos, CAD, and a physical inspection of the rover
- Organized and converted all CAD of the rover from Solidworks to Fusion 360 files in the MIL Fusion Team.
- Created a full assembly (with motion) of the Rover in Fusion 360 
- Discussed with Leaders and other project members about the overall goal of the rover project
    - We decided that we definitely want to do a chassis redesign

[image]

What's coming in the week of 2/3/25:

- Meet with other group member to fully establish outline/gameplan for the rover project
- Do further research on the rocker bogie mechanism for the future redesign of the chassis.
- Create napkin sketches of new Rover frame
- Discuss possible materials for new threads with other memebers.

GabeHoberman, 2025-02-02:
Week of 1/27/25 Update:
- Assisted in converting Fusion 360 Bodies into Components
- Read through prior rover documentation to understand its design.
- Began research on how to design new rover chassis (Thinking will still use Rocker-bogie suspension so the rover could be used for outdoor applications).
- Talked with leadership about project expectations.

Goals or next week:
- Ditto what Jeremiah said
- Discuss creating framing out of T-slotted Rail for frame to allow for modular design.

GabeHoberman, 2025-02-09:
Week of 2/3/25 Update:

- Generated napkins drawing of new rover frame.
- Continued research into rover design.
- Scheduled a meeting with the rover mechanical team for 2/11 to go over napkin drawings and solidify design.

Goals for next week:

- Meeting on 2/11.
- After meeting start generating CAD for frame design.

Danimata10, 2025-02-11:
Week of 2/3/25 Update:

- Meet with the group to establish a gameplan for the rover project.
- Made paper sketches of new rover frame.
- got up-to-date with the outcome of Week of 1/27/25.
- Set up a meeting on Tuesday to go over the napkins drawing.

Goals for next week:

- Attend 2/11 meeting to go over napkins drawings.
- Start working on getting the rover to move using an RC controller.
- Define specific tasks for each of the group members, so we can work more effectively.

GabeHoberman, 2025-02-16:
Week of 2/10 Update:

- Met with team and established general rover design.
- This week I created CAD for the Rover's Rocker Bogie suspension, dimensions rougly reflect the same general size as the prior rover.

[image]

Goals for next week:
- Work with the mechanical team to fully constrain the suspension and its connections to the chasis and wheels.

SaintFStudios, 2025-02-17:
Week of 2/10 Update:
- Met with the mech team and established a general new rover design
- Discussed possible upgrades to current rover swerve drive 
- Create prototype sketches of swerve drive
- Created a prototype CAD for an upgraded swerve drive meant for both new and current rover. It swaps out the DC motor for a 150 kg servo motor for positioning. 

[image]

Goals for next week:
- Meet with the mech team again to establish constraints on the suspension and its connections to the chassis and wheels.
- Further develop V1 upgraded swerve drive 
- Meet with the Electrical & Software team to talk to them about the possibility of upgrading to a high brushless torque motor to replace the DC drive motors.
- Research/Ask around about creating the optimal mounts for the motors

Danimata10, 2025-02-17:
Week of 2/10 Update:

- Met with the mech team and fully defined the new design.
- Assigned individual tasks for each team member.
- Started working on a new frame design based on the constraints given in the meeting.

Goals for next week:

- Meet with the Electrical & Software team to get information about the level of autonomy and use of sensors and actuators
- Further develop the frame 
- Meet with the group again to merge individual ideas

amcaleer271, 2025-02-17:
This is looking good so far! Make sure to keep in touch with the electrical and software people that are working on this.

GabeHoberman, 2025-02-23:
Week of 2/17 Update:
- Met with Electrical Team after the General Meeting, established a meeting with them for 2/25 to discuss how to improve old rover design and what we want to implement for version 2.
- Researched alternative forms of Rocker Bogie suspensions that utilized a spring for Shock Absorbing or Weight Balancing. Concluded that it might not be necessary for our use case.

Goal for next week:
- Focus on fixing the old version of the rover. Meet with the electrical team and decide what elements need to be replaced/improved.

Danimata10, 2025-02-24:
-Week of 2/17 Update:

-Further developed the frame design
- Inspected the rover and analyzed the problems regarding the constraintment of the back wheels
- Met with the electrical team to set up a meeting for 2/25. We plan to discuss individual ideas and developments as well as objectives for the new version of the rover.
- Discussed the posibility of using brushless motors with the electrical team. We also discussed the use of a lidar system to make the vehicle autonomous

Goal for next week:

- Fixing the old version and meet with the electrical team to elaborate a plan to create a new version of the rover that includes the constraints imposed by both teams
- Keep developing the frame individually

SaintFStudios, 2025-02-24:
Week of 2/17 Update:
-  Further developed V1 Swerve drive and current working on CAD to use FRC servos with 1:1 gear train. This modification moves the motor away from the center so that it doesn't bear the full weight of the rover and instead just transmits rotational force
- Team decided to keep current DC brushed motors as the drive motors for the old rover but using FRC motors for the new rover
- The Team went over possible changes to the chasis interior configuration that included adding more flat surface area to mount electronics  

Goal for next week:

- Further develop V1 upgraded swerve drive
- Meet with Mechanical and Electrical teams this week to talk about future changes/improvements to the rover

(Busy week because of exams and reports will have update later this week)

dylan-duenas, 2025-02-27:
Finalized azimuth design on Solidworks, getting BOM ready for export. The solution uses 1 axon servo (currently) geared 1:1 with 2 30T gobilda gears. The azimuth is driven by a 8mm live hex shaft to 2 hubs (1 top and 1 bottom), with a bearing on top and on the bottom pressed into a 3DP. The only things still missing are the mounting holes (to be confirmed with physical rover because not in the old CAD) and the heat set inserts to bolt into the azimuth assembly.

[image]

SaintFStudios, 2025-03-02:
This Week Update

- Created CAD for Drive Assembly that allows the replacement of the old 90:1 DC motors for New GoBilda Motors with embedded encoders 
- Designed new motor brackets (blue parts) along with a motor holder (bottom orange part) that holds the motor in place in the assembly.
- The brackets are CNC machined, and the holder is 3D printed with PETG
- Brackets attach to the holder with M4 screws with M4 heated inserts embedding into the holder.
- We still plan to use the original wheels from the older rover design. 
- Actuation bracket (top orange part) attaches to azimuth through GoBilda Hub attached to shaft

[image]

[image]

[image]

GabeHoberman, 2025-03-02:
This Week Update:
- Started CAD for a 3D printed part to help hold/organize electrical components within the Rover chassis. 
- Got associated measurements in lab this week.
- Will review with electrical engineering team to make sure that layout/spacing/placement is desirable for electronic components.

Danimata10, 2025-03-03:
Week of 2/24 Update:

- We meet up with tha electrical team and review their fixed version of the rover.
- In together with the Electrical team, we defined what will the next steps for the rover be.
- Started finding solutions for making the wheel's holes smaller in case the motors are changed.

Goal for next week:

- Meet with Mechanical and Electrical teams this week to talk about future changes/improvements to the rover.
- Further develop rover V1.
- Help with the development of a better axis constraint system at the wheels.

GabeHoberman, 2025-03-09:
Week of 3/3 Update:

- Made electrical component mount for Motor Controller and Position Terminal (Uploaded into "Fixed Rover CAD")
- Designed option to use mounting pegs or M3 fasteners. Going to get feedback from Electrical during our Tuesday meeting with the intension of printing/mounting soon.

[image]

[image]

[image]

[image]

SaintFStudios, 2025-03-10:
Week of 3/3 Update:

- Continued to work on the new drive assembly for the rover
- Found 2 main problems with using the gobilda motors: 1: the shaft diameter of the Gobilda motor is smaller than the mounting hole on the wheel; 2: there is not enough clearance between the wheel and the motor in the real-life assembly. 
- The second I believed I solved by modifying the shape of the motor holder to place the motor further back and I'm planning to use countersunk screws to provide greater clearance. 
- 3D printed a mock-up of the drive assembly to get an idea of how it looks in real-life
- Found a possible separate problem with Gobilda motors: the diameter of the motor seems to be a little too big to fit into the current mounting place of the middle wheels of the rover. We don't want to replace the mount because it is a complex part that is welded together.
-A possible solution to the above problem is to file down the sides of the motor so that it can fit inside the current mount.

Next Week: 
- Meet with mechanical to discuss motor shaft diameter issue
- Continue work on drive assembly

V2 mock up
[image]

[image]

[image]

V2 CAD

[image]

[image]

[image]

Clearance issues from V1

[image]

Shaft Dia issue

[image]

Danimata10, 2025-03-10:
Week 3/3 update

- Not much as I was full with exams, but attended the weekly meeting.

-Brainstormed about solutions for diameter fitting in the motor shaft.

- discussed the updates regarding the new design

As for next week:

- Find a new part to be fixed or created

- schedule a meeting with electrical

dylan-duenas, 2025-03-10:
Adjusted the mount to fit heat-set inserts, going to try to print a copy to look at for tomorrow

[image]

amcaleer271, 2025-03-13:
I printed 3 of the motor driver mounts so Joe can test the rover.

GabeHoberman, 2025-03-16:
3/10 Update:

- Joe sent me the specifications for a board Russel is working on. I made a mount for the specifications he sent me. It is saved under the "Fixed Rover CAD" folder as "Russel's Board Mount".
- The board connects to the Rover with M3 Fasteners
- I want to try to see if these mounting pegs are stable enough when 3D printed to hold. When I get back from Spring Break I will print off a version to see how it performs.

[image]

SaintFStudios, 2025-03-17:
3/10 Update:

-  Made small adjustments to the Drive Assembly CAD to improve its real-life accuracy
- Created a 3D-Printed Spacer for the Wheel Hub to allow for smaller diameter GoBilda Motor shaft to connect properly with the wheel
- Met with other members of the Mech team to review our progress; will start getting BOM ready for drive Assembly frame after the break

[image]
[image]

Danimata10, 2025-03-17:
Week of 3/10 update:

- After researching about ways to fix the clearance problem in the motor shaft, the most optimal way is using shaft couplings. I want to design one to 3D print and test if a plastic one works. However, I think that using a plastic piece to sit between metal pieces will not work as the plastic piece would significantly wear over a small period of time, making it a provisional but not final solution. If it does not work, I found a company that sells metal coupling specifically designed for this problem.

Goals for after spring break

- Have the 3D piece printed (i'll use my house's printer, as the lab will probably be closed over the break)

- Meet with the rover team to present this opinion and make a decision.

- Get a price quote for the OTS couplings

Danimata10, 2025-03-17:
- Just saw that Jeremiah did a piece to solve the problem and is going to test it. I'll probably do my version and we'll see if any of those work and which one works better. I will still propose the OTS idea as I think wear will hugely impact any plastic piece that tries to do that job. At the moment, let's  just test it and decide. 🫡

dylan-duenas, 2025-03-27:
Re-CADed the previous motor mount in fusion to better accommodate the new motor mounting for the GoBilda motors.
[image]

Since the motor mount is meant to attach only on the bottom, as opposed to the previous top and bottom mount, it is a much lower profile and robust solution with thicker walls and a thicker top section.
[image]

Here, the azimuth motor is still geared 1:1 with the output because software wanted to use the absolute encoder onboard the servo. This still feels a little bit sketchy (given most similar scale FTC modules usually use axon max+ instead of axon minis), but still seems possible given the moderate load these motors will be under. Plus, since the drive shaft is separate from the motor spline, the actual servo horn shouldn't directly experience the shock loads in the system.
[image]

I used M3 heat-set inserts (the same ones as used in the drive motor mount to save cost and needless complexity) for the mounting, which may have to be revisited later because of the cheese-holed chassis rails.
[image]

SaintFStudios, 2025-03-30:
Week 2/24 Update

- Combined Azimuth Assembly with the drive assembly with minor edits to the motor brackets.
- Working towards to creating new full assembly with new drive motors, motor brackets, motor holder cases, etc.
- Started filling out BOM Excel 
- [Rover BOM](https://uflorida-my.sharepoint.com/:x:/g/personal/jeremiahstfleur_ufl_edu/EWjwfq3xYXlNuBbTT8w2kCEBB1WcbbOEePsyvjAa4-wFQw?e=AzTzDW)
- Printed and Tested 3D printed Wheel Hub spacers 

[image]

[image]

[image]

GabeHoberman, 2025-03-30:
Week 3/24 Update

- Wasn't able to do much this week as a result of exams and assignments.
- Added fasteners and nuts required for electronic mounts into BOM. I think we can get away with just threading fastener through the print and connecting it to frame with a nut. Will check to make sure this is secure enough in lab this week.
- This week during our rover meeting, I am going to get feedback from electrical mounts and make adjustments if needed. Will begin seeking out a new task this week, as mechanical requirements for the rover are wrapping up.

GabeHoberman, 2025-04-06:
Week 3/31 Update

- The Motor Controller Mount's mounting peg locations were slightly too far apart. I readjusted them to get a better fit.
- Looked into changing the material of the Chassis, Adam recommended using Aluminum.
- I broke up the current chassis into sections to see how much the conversion into an alluminum chassis would cost. Currently the chassis is 0.1 inches thick. I assumed that we would want to connect the Aluminum paneling with fasteners (didn't know if we had anyone who can weld well). If we wanted to connect the panels with fasteners I opted to approximate costs with a 1/4" thick aluminum sheet.
- If we were to make the current Chassis design out of 6061 Aluminum, I estimate that it would cost around $177.14 using a 12"x48" sheet and a 12"x24". If we were able to fabricate it with a single 12"x72" sheet it would cost around $141.71.
- My work is attached as a pdf, if you want to see how I broke it down

[Aluminum Panel Cost Calc.pdf](https://github.com/user-attachments/files/19622171/Aluminum.Panel.Cost.Calc.pdf)

dylan-duenas, 2025-04-07:
I resubmitted the current azimuth mount print to the print queue because of some earlier issues (my slicer didn't autogenerate the supports). I'm also considering some edits to better accommodate the mounting on the existing cheese-holed chassis. I added an extra spacer aligned with the hole pattern of the chassis, and also edited the C-C of the heat set inserts to match the C-C of the chassis. Also got rid of the middle insert, and am predicting that 2 bolts on each side is sufficient.

Here's the new assembly with the spacer incorporated
[image]

And here's the spacer itself.
[image]

Additionally, lets talk about the chassis re-construction on Tuesday so we can lock that in and start designing some ideas!

Danimata10, 2025-04-07:
3/31 update

- Went over the BOM and checked everything was fine
- Discussed with some members about the re-design of chassis as well as material selection
- Started researching on cool ideas to improve chassis selection 

For next week:

-Start doing some CAD designs for de chassis and meet with the group to gather ideas

GabeHoberman, 2025-04-13:
4/7 Update

- Decided to connect electronic mounts to rover using Velcro strips instead of fasteners.
- Removed fastener connections on CAD and made the mounts thinner
- Learned how to 3D print and generate G-Code
- Added Prototypes into the print que, if they appear like they will work during our Tuesday meeting will print off the remainder needed.

dylan-duenas, 2025-04-14:
Designed a sheet metal body for the rover, but ultimately decided against it because the rover would need a more significant redesign if it were designed with a mechanism/payload in mind. Working on a requirements design document for any future potential rover iteration if there's any interest in continuing the rover project past just a drivetrain.

[image]

Danimata10, 2025-04-14:
This week:

I have been working on the base for future attachments, I am doing the CAD design in fusion under the new CAD design file. I plan to have it done by Tuesday to show it as an option to expand the Rover project for next semesters.

Next Week:

Finishing the CAD design and meet with the group to keep the good work on the rover

GabeHoberman, 2025-04-21:
This week: 
- Wasn't able to do much due to exams and holiday weekend.
- Spent some time reading through the MILPedia, learning how to create a page and edit it. Planning on spending time this week adding documentation about the rover project to the wiki.

Danimata10, 2025-04-21:
This week:

* Finished the CAD design for the chassis, which would open new projects for the rover. I showed it in our weekly meeting and received feedback.

Next week:

* Apply the changes in the design.

closed 2025-08-21 by amcaleer271