uf-mil/mechanical #46: NaviGator Frame

amcaleer271, 2025-01-28:
A frame to house the electronic boxes and sensors for NaviGator. The plan is to build this out of 80-20 so it is easy to expand.

amcaleer271, 2025-01-28:
An initial design for the frame has been created in Fusion (img below). We are going to wait on the new electronics boxes to arrive before designing the mechanism to mount the boxes, as there is no CAD we can find for the boxes.

[image]

space-cat-explorer, 2025-01-29:
Added myself to help with the construction of frame. Link any parts you wanna use to create the latches. I will also look around for stuff.

amcaleer271, 2025-01-30:
I realized a flaw in the current design I had where some parts were only held on by the frictional forces between the t-nuts and the aluminum extrusion. To fix this I've started remaking the frame using thru bolts with shanks, this way for the system to fail, multiple bolts would have to shear in half.

amcaleer271, 2025-01-30:
Updated frame!

[image]

amcaleer271, 2025-02-13:
The new boxes have arrived, so we're going to model them in fusion. @space-cat-explorer suggested using a clamping system to hold the boxes in place, so I'm working on finding clamps and a way to mount them to the 80-20. So far I'm liking clamps like [these ](https://www.mcmaster.com/products/toggle-clamps/low-profile-hold-down-toggle-clamps-9/). 

@nonagaga and I are going to work on modeling the new boxes accurately, then I can finish the frame with the mounting system.

nonagaga, 2025-03-17:
@amcaleer271 and I made use of the new WEN drill press we ordered, and have improved tolerances on the drilled 80/20.

Using the drill press, we expect to finish the frame in 2 weeks (we're on spring break, for any future readers)

amcaleer271, 2025-04-03:
Built the superstructure of the frame and put bars running along the length of the bottom to hold the boxes. Need to find/get more of the t-slot inserts to finish it.

amcaleer271, 2025-08-25:
This semester we need to finish the frame ASAP, the boxes are almost ready to go on the boat! 

Remaining to be done:
* Attach frame to boat (probably via running some 80-20 along the top and drilling into the top plate of the boat, don't want to mess with the wire guards on the sides)
* Make the containment areas for the boxes
* Add cable runs between the boxes
* Add protection from water for the boxes

amcaleer271, 2025-10-07:
A few updates:
* The frame is built, it looks awesome and is very sturdy! We test fit a box and it's all good :)
* Working on retention brackets for the boxes. I've designed and am building doors, and also am going to add clamps to the sides to secure the boxes
* After that I want to add a "safety line" to attach the actual box to the top of the frame, just with paracord or something as a last line of defense.
* Then it'll be done