uf-mil/mechanical #48: [Structural] New Battery Mounts

amcaleer271, 2025-01-30:
We need a way to mount the batteries onto the new boat. The previous mount was welded on, which worked, but it would be nice for the mount to be removable so if we get different batteries or want to move where they are mounted we have that option.

adamhamdan8, 2025-01-30:
This might be hard to do, lets look at how other teams do theirs so we can find a semi-permanent method to attach them without taking the boat apart

RyanHoburg, 2025-01-30:
[image]

RyanHoburg, 2025-01-30:
If these brackets aren't doing anything I think it would be relatively easy to bolt on another plate that I can drill through

amcaleer271, 2025-01-31:
I don't think they're doing anything so go ahead and do that!

RyanHoburg, 2025-02-04:
[image]
Looking online it seems like a good amount of boats just strap their batteries down like the picture above

RyanHoburg, 2025-02-04:
[image]
The battery fits well in one of the existing brackets and I don't think it would fall off if we just sat the battery in there. To be safe, I think that straps would be plenty to be sure that the battery is mounted safely and they would be easy to run through the existing holes and easy to remove. The plate on the other side of the boat won't work so I think I'll just copy the existing bracket assembly over to the other side.

RyanHoburg, 2025-02-04:
[image]
This is the bracket assembly, I haven't added it to the main assembly yet

RyanHoburg, 2025-02-13:
[image]

RyanHoburg, 2025-02-13:
https://www.uscargocontrol.com/products/1-x-6-Retractable-Ratchet-Strap-w-Push-Button-Release-4-PK?gclid=EAIaIQobChMI1cjdwZHC7gIVwbyGCh3WowT6EAYYASABEgLeb_D_BwE

RyanHoburg, 2025-02-20:
Made drawings for the parts, would be great if @amcaleer271 or someone good with drawings could take a look because Im pretty sure i'm missing something

amcaleer271, 2025-02-20:
The metal has been ordered, we went with 2 0.125" 12"x36" aluminum sheets. Once we have a delivery date we'll make an appt with the fab lab to get them waterjet cut.

RyanHoburg, 2025-02-21:
Added MilPedia page

adamhamdan8, 2025-03-06:
@amcaleer271 Got the parts waterjet today! Came out looking great

RyanHoburg, 2025-03-07:
[image]

amcaleer271, 2025-03-07:
The new mounts look great, nice work!

Closing issue

closed 2025-03-07 by amcaleer271

RyanHoburg, 2025-03-30:
Decided to go back and add the L brackets and fasteners so there is 0 chance this thing can move, all order form stuff and a little more metal grinding.

RyanHoburg, 2025-04-19:
[image]

Looks great all bolted together