uf-mil/mechanical #51: SubjuGator Props

amcaleer271, 2025-02-10:


amcaleer271, 2025-02-21:
Just to keep track, Lucas is working on the Nav markers, Ali on the path markers, and Lana on the start gate. 

For the start gate and path markers, some dimensions are available [here](https://robonation.org/app/uploads/sites/4/2024/07/2024-RoboSub_Team-Handbook_v2.pdf) . Scroll down to the task descriptions and you should be able to find them. Unfortunately as the navigation channel is new this year, all we have to go off so far is the pretty undescriptive 2025 Task Ideas (linked in the original issue above)

askarshinskifred, 2025-02-27:
I asked Adam all my questions, here's the answers:
Existing size of PVC pipes: 1" inner diameter, 1.25" outer diameter
Height of path markers: No specific height necessary, old model was approximately 9" from the top of the horizontal PVC to the top of the vertical PVC
Nut and bolt to rotate: use a lock tight nut, has nylon to prevent it from coming off but still allows for rotation
Keep PVC from coming apart: PVC cement is good to glue the PVC pipes together
 Weighted: Find small diving weights that can fit inside the PVC, you can insert the weights when you unscrew the PVC from the base
McMaster components: are fine to use as long as they are not super big (then it gets expensive)

Also note to self: Add caps on the ends of PVC like the old model to make it more stable. Right now, the middle is the only part in contact with the ground.

l-gerstenberger, 2025-03-02:
Spoke to Adam at the meeting on Thursday and I need to switch all the PVC to 1", add weights to the plastic markers in the middle of the start gate, and re-drill the new PVC that comes in and add weights to those. He mentioned that using threader 90-degree fittings would help assembly and taking it apart in the water. I have found some PVC options on Home Depot (elbow fitting, 10 ft, 5 ft) that I believe will work. I can fill out an order form for these parts once they are confirmed to work and then get started on re-assembling the gate with the new pieces.
https://www.homedepot.com/p/Charlotte-Pipe-1-in-x-10-ft-Plastic-Plain-End-Pipe-PVC200100600/100348483
https://www.homedepot.com/p/Cantex-1-in-PVC-Male-Terminal-Adapter-Conduit-Fitting-for-Cantex-PVC-Conduits-R5140105/202043379
https://www.homedepot.com/p/Charlotte-Pipe-1-in-Schedule-40-PVC-90-Degree-MPT-x-FPT-Street-Elbow-Fitting-PVC023071000HD/204837327
https://www.homedepot.com/p/Apollo-1-in-x-5-ft-White-PEX-B-Pipe-APPW51/301541233

lucas-reilly, 2025-03-03:
Working on nav markers. Missed the meeting on Thursday but will continue building with newly purchased colored plastic markers and pool noodles. Have an individualized and cloneable design in mind that is held down by an adjustable rope attached to the top and bottom of a pool using a weighted system. Might need to purchase more supplies to copy the design when the first markers are made.

askarshinskifred, 2025-03-06:
Talked to Adam, here's some notes:

- Calculate the buoyant force and weight so that mg >> Fb
- Drill a hole in the PVC to get the air out
- Using size 1 PVC (best matches the existing PVC on the old model, Adam said if they aren't compatible he can get it from Lowes

Here's the new sources:
- [Straight PVC Pipe](https://www.mcmaster.com/catalog/131/114/48925K93) (48925K93)
- [Tee Connector](https://www.mcmaster.com/catalog/131/112/4880K43) (4880K43)
- [Threaded adaptor](https://www.mcmaster.com/catalog/131/114/4880K63) (4880K63)
- [Fixed Female Threaded Pipe Flanges](https://www.mcmaster.com/catalog/131/123/4596K154) (4596K154)
- [0.75" diameter weights to put in PVC](https://www.amazon.com/ECOFT-Raindrop-Assorted-Saltwater-Freshwater/dp/B08PVD8PC4/ref=sr_1_26?crid=OF5T2D2FSWOQ&dib=eyJ2IjoiMSJ9.7d6LAHFGd4rDJcNL22-RuylUmrnRLVGryUCcIhfaflGpBz0S8I4YdsjhSvHntBxY2xFVuefIyGT0SVZ_POXicHxtx4DN0ERV81I2rcVP6oWALlSjmG0OYqPxcv8zsHQDwdFL7taVCOA-Ncr_uwCrgjxED4wUMzJKobfSAy8_kxJfjEv0Akov7IiD4SSv0ECX0uemu50kHUvzmoNd9Mglw0aV6mJbDVi5xv0MWbBtnaRyzX7sgBDnHlEUWOzlgzyCdRlfYruUObLysMPFS0oaGDVS-pPS94VFXoNy1IgrcB_F7PvwNDKniBV9dBBnvQaCVcHtBZ1q0leqBgLgj3JrWTiBucg-MmAXK0S5gGYzeRStodgKi9HrK1fDpiqY1-NaWiyQ78F-5isbfy1OKJ_spNCJci5G0RHRF8E0LeYvH8XCJQG3kSISbM7dDAe3skFt.d_6xNASvtoIT6a7fnuxXdfBzHJ74Sp6RKa5mZhuEXK4&dib_tag=se&keywords=cylindrical%2Bthin%2Bweights&qid=1741284656&sprefix=cylindrical%2Bthin%2Bweights%2Caps%2C122&sr=8-26&th=1)
- [Black-Corrosion-Resistant-Coated Steel Nylon-Inserted Locknuts](https://www.mcmaster.com/catalog/131/3658/91333A150) (91333a150)

I have worked on updating the Fusion model and will continue to complete it. 
Note to self: use 1/4" - 20 for bolt/nut size

askarshinskifred, 2025-03-07:
I have finished the final assembly in Fusion.
[image]
[image]

I applied the proper material to all the components and analyzed the assembly. The total volume is 201.6 in^3 and total mass is 143.794 ounces. Converting these to m^3 and kg, I calculated the buoyant force to be 32.278 N and the weight of the path markers to be 40 N. Although these calculations are approximations, the weight of the path markers should be greater than the buoyant force, meaning they will stay submerged on their own. However, if necessary the fishing weights can be placed inside of the PVC to ensure this. 

Next week, I will go over my design with one of the leads, order parts, and write the MilPedia page for the path markers.

l-gerstenberger, 2025-03-09:
Submitted an order form for the PVC and elbow fittings. Might use a PVC threader to attach the fittings. Once the parts come in, I will replace all the PVC on the original start gate and calculate how much extra weight is needed for the center of the gate. I will also need to redrill all the PVC and reattach the cardboard red and blue pieces.

askarshinskifred, 2025-03-13:
I will also order washers for the [path markers ](https://www.mcmaster.com/catalog/131/3682/92141A029)(92141A029) and [screws ](https://www.mcmaster.com/catalog/131/3446/92949A548)(92949A548)

l-gerstenberger, 2025-03-16:
Was waiting for the PVC to arrive, but I looked at it in the shop this week and determined how I will attach the pieces. Once I get back from break, I will get started on that.

askarshinskifred, 2025-03-25:
I created the MilPedia page for the path markers! I just need to figure out how to add the sidebar, which I'll ask about at Thursday's meeting. 
[image]

l-gerstenberger, 2025-03-31:
Wasn't able to do much this weekend, but I plan to go to the lab on Monday to assemble the pieces.

l-gerstenberger, 2025-04-06:
Went into the lab on Monday to make sure all the PVC look correct. Going to the lab on Tuesday to assemble everything if time permits.

askarshinskifred, 2025-04-08:
I went to the lab today and found all the materials I ordered. The fishing weights seem like they will work great! I cut the PVC pipes, but may cut them down more later (I didn't want to accidentally cut too much). I forgot that the path markers need to be orange, so I ordered [orange spray paint](https://www.lowes.com/pd/Rust-Oleum-2X-Ultra-Cover-Gloss-Real-Orange-Spray-Paint-and-Primer-In-One-NET-WT-12-oz/5014069569) that is also a primer. I marked on the wood with pencil the size of the path marker for cutting. I am hoping to cut the boards this or next week, and then all that will be left is to use PVC cement on the PVC and finish assembling everything. I am going to arrange a time with one of the leads to help with cutting the board because I am not confident doing it by myself!

adamhamdan8, 2025-04-09:
@askarshinskifred Just spoke with Daniel and he said the pvc can stick out the sides, but he prefers the height to be shorter. The reason is that if we want to take large ranges of data its better if its deeper in the pool. We can get closer to make the path marker to appear bigger but we can only go so far away. Great work so far!

askarshinskifred, 2025-04-10:
I cut the wood into rectangular pieces, but I still need to make the edges curved. I also cut down the PVC a bit more, making one 1/2in taller than the other to account for one path marker resting on the other! In the following weeks I need to drill the holes in the wood and apply orange paint.

l-gerstenberger, 2025-04-14:
I cut the longer pvc into two equal lengths so that it is easier to assemble in the water. I spoke to Adam about using threaded elbow pieces vs the ones that were used previously, and I will try out the threaded first. I still need to drill the ends of two of the pvcs and then just put everything together and add the fishing weights to the images. I plan on going in on Tuesday to drill the holes. I found that a 6/32 x 2" should work to attach the elbow pieces.

lucas-reilly, 2025-04-14:
Ordered red PVC for Nav markers instead of having to deal with plastic wrap, using white PVC for other markers. The simple design consists of having each nav marker independent of one another so spacing is easier to change. There will be a weight at the bottom of each and a rope connected to the top via drilled holes and a simple knot. Once I have the first pair of markers assembed, I will order the rest of the pvc.

askarshinskifred, 2025-04-15:
I just applied cement and primer to the PVC. At this week's meeting I need to ask about cementing and priming the part with the threads, because the flanges and the threaded adaptor aren't perfect and I don't want to permanently fix their orientation yet. All that is left now is to paint the path markers orange, cut the rounded edges, and drill holes through the wood and PVC!

l-gerstenberger, 2025-04-21:
Did not have much time this week with final projects but plan on going in during reading days to finish up assembly.

l-gerstenberger, 2025-04-22:
Final Report: The start gate that MIL was working with for the RoboSub previously tended to bend and the images hung at the center would float to the surface. To solve this issue, I replaced the ½” PVC to 1” PVC. Furthermore, I decided that it would be more beneficial to use threaded elbow brackets to attach the various pieces, as I figured it would be easier to assemble and dismantle between testing. To do so, the PVC will need to be threaded to fit into the new elbow brackets. To solve the images floating to the surface, I decided to add fishing weights to the bottom of the images. While all of the pieces have been ordered and the PVC cut to size, I did not have the chance to assemble the entire new start gate before the semester ended. I intend to visit the shop in the next week to drill the PVC and add the fishing weights, however, I might not have time to put all of the pieces together before the semester ends.

closed 2025-08-21 by amcaleer271