uf-mil/mechanical #52: SubjuGator Props

amcaleer271, 2025-02-10:


tylergunther111, 2025-03-05:
I can help out with this if you guys give me the rundown at thursday's meeting

tylergunther111, 2025-03-30:
[image]

Remodeled nav tube bracket. New part is now a fusion native file, so it can be modified. The inner diameter of the new bracket is the same as the outer diameter of the nav tube, whereas the prior version was a larger diameter with foam tape between the bracket and tube. The center piece will be attached to the nav tube, either with a hose clamp or epoxy. It fits into the notches on either bracket to constrain rotation and translation of the tube.

[image]

I brought the new nav tube subassembly into the Sub9 CAD but the file is kinda messed up from solidworks. I had to capture design history to get fusion to let me use the align command and create rigid groups, but I got everything in place. Next week, I'll print the new brackets and see if they work out.

tylergunther111, 2025-04-05:
Printed and fitted everything this week. The alignment tab has a little bit of play so I'm going to print re print it with a tighter fit next week. Once that gets printed, we just have to epoxy the alignment tab to the tube.

tylergunther111, 2025-04-13:
updated alignment tab is sliced and waiting to be printed. I brought the brackets and nav tube into the new CAD model.

closed 2025-04-17 by amcaleer271

tylergunther111, 2025-04-17:
Finished gluing the new alignment tab to the nav tube. Project is complete and Sub9 CAD has been updated to reflect the changes.