uf-mil/mechanical #54: NaviGator Thruster Shrouds

amcaleer271, 2025-02-10:


BryanBoc, 2025-02-23:
Started research and also was able to find the thruster and old shroud at Cimar.

RyanHoburg, 2025-02-26:
[image]

RyanHoburg, 2025-02-26:
https://tacklebox.co.uk/boating-accessories/tb-all-stainless-steel-prop-guard/

RyanHoburg, 2025-02-26:
https://www.amazon.com/Propeller-Propellers-Protector-Stainless-Prevention/dp/B0C3GM38DX

RyanHoburg, 2025-03-10:
The linked shrouds above will work for the two old thrusters we have. I think the Tackle Box brand shroud looks higher quality and 
 it would cover the prop to the distance we need. I'm not sure if the solid metal band part would be too much drag through the water though. The price has also gone up $20 since I first viewed it.

RyanHoburg, 2025-03-10:
[image]
The new thrusters don't have a round shaft we can mount on but I wonder if we might be able to use a more traditional prop guard like the one above.

RyanHoburg, 2025-03-10:
@BryanBoc and I took a look at the new thrusters and there probably isn't going to be an easy way to mount a shroud. So far can't find anything online that would work.

[image]

They have a longer profile for less drag through the water, more brainstorming needed.

RyanHoburg, 2025-03-13:
Reached out to Torqeedo to see if they have any ideas

RyanHoburg, 2025-03-30:
We have two ideas for attaching a shroud to the torquedo thruster. One involves clamping onto the main body going down and attaching a traditional prop guard to the clamp. 3D printing the clamp for proof of concept. The other idea is using hose clamps or sheet metal strips to wrap around the round in which the propellor rotates and using more wire/strips to create a guard. I think we will get started on both ideas and see which works better.

BryanBoc, 2025-03-31:
This is what I envision for the latter idea. We can wrap aluminum strips around the front and back of the bottom of the thruster. Note that they're gonna have to take on the the conical shape of the bottom part since its not a cylinder. If we connect them with a strip on either side that should constrain the structure enough so it doesn't budge. Then we can weld (or use fasteners to secure) the actual shroud part which could just be a wire frame similar to the previous one.

[image]

RyanHoburg, 2025-04-01:
[image]

RyanHoburg, 2025-04-01:
https://www.amazon.com/SupremeMarine-Propeller-Polypropylene-9-9-120-Outboard/dp/B0D56PHQGT?th=1
https://a.co/d/fkJJG91
https://a.co/d/f16aoQF
https://www.mcmaster.com/products/hose-clamps/hose-clamps-2~/

RyanHoburg, 2025-04-06:
Waiting on ordered parts. We are trying to wrap the guard around the thruster housing first. If this works we will order the parts to complete the other thruster. If it doesn't we will move on to our concept that mounts to the body extending down to the thruster housing.

BryanBoc, 2025-04-07:
^^

RyanHoburg, 2025-04-14:
^^

amcaleer271, 2025-08-25:
We're going to mount the old thrusters first so we're going to put this on hold and work on shrouds for the old thrusters