uf-mil/mechanical #56: Fan splash guards

amcaleer271, 2025-02-10:
We need a way to make sure water doesn't get into the new electronics boxes.

raffimadalian10, 2025-03-17:
-  Started researching ways in order to protect the fans from any interface with water but still allow for airflow in order for the fans to be able to function. 
- One option: Baffled Air vents: labyrinth-style venting where air can pass but water gets trapped or redirected away
- Another option: PTFE membranes, like Gore-Tex, which allow air to pass through but block water.

raffimadalian10, 2025-03-31:
https://northspore.com/products/adhesive-ptfe-filters-for-3-monotub-holes

Found these high airflow, ultra-thin filters which can cover the vent fan openings -- 3 inch in diameter.

amcaleer271, 2025-04-03:
They look good, just verify they fit over the fan holes

raffimadalian10, 2025-04-21:
Update: Waiting on holes to be cut in the pelican boxes to verify these filters will cover effectively.