uf-mil/mechanical #57: NaviGator Cable Runs

amcaleer271, 2025-02-17:
The cables going between the pelican boxes on NaviGator need runs to keep them organized and safe. 

The cable runs can lay on top of the 80-20 rails at the bottom of the NaviGator frame, but should be secured with bolts. 

Below is an image from below NaviGator showing where cable runs are needed. These should be channels where cables can lay and be secured. Additionally, there should be a way to cover the runs in the event of bad weather. 

[image]

amcaleer271, 2025-02-17:
Some options are to use PVC or a similar tubing system. Avoid manufacturing where you don't need to!

raffimadalian10, 2025-02-22:
Researched different options of easily accessible covering for cables. Came to the conclusion that a PVC tubing system would be the best option. Would have PVC tubes cut in half horizontally connected via Tee-connectors and have a slightly larger PVC pipes with a cutout that can freely rotate around the latter PVC tubes described. Illustrated below. 

[image]

raffimadalian10, 2025-03-01:
[image]

- Finished assembly of cable runs - now need to see how it will fit in the full assembly. May need to wait until the pelican boxes are in the assembly to do this.

raffimadalian10, 2025-03-10:
- Consolidated design to accommodate for only two pelican boxes instead of 4. Edited dimensions to tailor the spacing between both boxes.

raffimadalian10, 2025-03-31:
- https://www.mcmaster.com/48925K304

- https://www.mcmaster.com/48925K305

- Here are the two PVC pipes needed for the design

raffimadalian10, 2025-03-31:
Also question -- should I make holes in the pelican boxes to let the PVC pipe design run through them?

amcaleer271, 2025-04-03:
We're making a panel that will bring the cables out of the boxes, so the PVC doesn't need to go into the boxes.

raffimadalian10, 2025-04-07:
[image]
[image]

- added cable runs to full assembly according to how we discussed after mech meeting. I know the 80/20 structure is gonna change but I believe we should be able to secure the pvc to the 80/20 by using hose clamps. If we can get the pvc from lowes I can start working on it this week.

raffimadalian10, 2025-04-21:
Update: We are waiting on the frame and pelican boxes to be mounted to go to Lowe's, get PVC and hose clamps, and assemble according to plan.

amcaleer271, 2026-01-03:
To get NaviGator running we ran all the cables along the underside of the top plate with no issues. With NaviGator not being used for competition we'll close this issue