uf-mil/mechanical #58: Sub 9 Thruster Potting and Mounting

adamhamdan8, 2025-02-27:
The sub9 thrusters need to be tested, potted, and mounted in proper locations so that the cable runs are good. Keep in mind that they should be labeled so that we know which thruster goes where easily.

They first need to be mounted for some events before we can test, pott, label, and permanently mount them!

RyanHoburg, 2025-02-28:
Thrusters mounted for show, still need to measure and pot cables.

adamhamdan8, 2025-03-03:
Adrian just crimped some andersons to a esc and wetcon so that we can test the thrusters tomorrow. We will be looking at the quality of the solders too before we pot, and then we will mass pott before Friday. I have to print the shells before that as well!

adamhamdan8, 2025-03-06:
We tested one yesterday and it worked! Just going to test the rest of them, checking the soldering on the questionable ones just to make sure!

adamhamdan8, 2025-03-12:
Oh whoops I forgot to mention this but I did this last weekened, bang they are all on the sub now! I only included images of the potting but you can peep it on the actual sub

[image]
[image]

closed 2025-03-12 by adamhamdan8