uf-mil/mechanical #59: Sub9 Kill System

adamhamdan8, 2025-02-27:
The mechanism required to kill and unkill Sub9. We haven't yet decided if its going to be a switch, button, wand, or plunger type design. The mechanism, mounting system, magnet, and interface with electrical all need to be decided.

WestinZ, 2025-02-27:
As discussed with lorant, here are 2 Reed switches. You want to use Normally Closed (I think) which are SPST-NC. This should mean that the circuit breaks open with the magnets removed from the switch. 

https://www.digikey.com/en/products/detail/littelfuse-inc/59025-1-S-02-A/822381?gclsrc=aw.ds&&utm_adgroup=General&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_Zombie%20SKUs&utm_term=&utm_content=General&utm_id=go_cmp-17815035045_adg-_ad-__dev-c_ext-_prd-822381_sig-EAIaIQobChMIotyimMvkiwMV99oWBR1diQCoEAQYAiABEgJf0_D_BwE&gad_source=1&gclid=EAIaIQobChMIotyimMvkiwMV99oWBR1diQCoEAQYAiABEgJf0_D_BwE&gclsrc=aw.ds

https://www.digikey.com/en/products/detail/littelfuse-inc/59143-1-T-05-A/20482178?gclsrc=aw.ds&&utm_adgroup=General&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_Zombie%20SKUs&utm_term=&utm_content=General&utm_id=go_cmp-17815035045_adg-_ad-__dev-c_ext-_prd-20482178_sig-EAIaIQobChMIotyimMvkiwMV99oWBR1diQCoEAQYBCABEgJ9DPD_BwE&gad_source=1&gclid=EAIaIQobChMIotyimMvkiwMV99oWBR1diQCoEAQYBCABEgJ9DPD_BwE&gclsrc=aw.ds

WestinZ, 2025-03-03:
**NORMALLY CLOSED SWITCHES**
"Cylinder" style switch https://www.digikey.com/en/products/detail/littelfuse-inc/59025-4-T-02-F/4778048
"Cylinder" style actuator https://www.digikey.com/en/products/detail/littelfuse-inc/57025-000/43961

"Box" style switch https://www.digikey.com/en/products/detail/littelfuse-inc/59140-040/412710
"Box" style actuator https://www.digikey.com/en/products/detail/littelfuse-inc/57140-000/43978

lorantdomokos, 2025-04-07:
@WestinZ Could you find these in normally open variety?

WestinZ, 2025-04-11:
"Cylinder" style switch NORMALLY OPEN https://www.digikey.com/en/products/detail/littelfuse-inc/59025-010/36970
"Cylinder" style actuator https://www.digikey.com/en/products/detail/littelfuse-inc/57025-000/43961 

"Box" style switch NORMALLY OPEN https://www.digikey.com/en/products/detail/littelfuse-inc/59140-010/43977 
"Box" style actuator https://www.digikey.com/en/products/detail/littelfuse-inc/57140-000/43978

@lorantdomokos

lorantdomokos, 2025-04-14:
@MjAdrian and I tested the NC switches and they worked but would have required an inverter. I just ordered the NO ones which should work.

amcaleer271, 2025-05-31:
Kill wand and system are on the sub, testing tomorrow!

amcaleer271, 2025-06-03:
The kill system was tested on Sunday and works great! I printed some spares of the end piece, as while it is easy to reglue the magnet, it does take some time. Closing issue!