uf-mil/mechanical #65: [Structural] PSU Mount for Navigator

nonagaga, 2025-04-10:
Current Navigator PSU mount needs to be adapted for new vertical mounting orientation and sufficiently reinforced to prevent failure.

nonagaga, 2025-04-13:
[image]

I was able to recreate the current 3D printed PSU bracket in Fusion, with some added support for the vertical orientation. I also intend to use the iron shelf brackets @space-cat-explorer (Sophie) ordered to further reinforce the design.

Next steps are to place an order for the specific 6-32 fasteners used for securing the PSU and to machine the iron bar if needed.

nonagaga, 2025-04-20:
First test print of the PSU mount has been submitted, looking to confirm fitment Monday next week and then print the real deal before this Wednesday.

nonagaga, 2025-04-30:
PSU mount has been finalized before leaving for break. The production part is attached to the PSU as shown. Heat set inserts will be used to fasten retaining bolts to the L bracket supports to make sure the assembly doesn't slide off. [image][image][image]

amcaleer271, 2025-05-01:
Is this ready to go in the box? If so we can close the issue

amcaleer271, 2025-08-25:
Just need to drill the L brackets into the base plate and slot it in!

nonagaga, 2025-09-19:
Holes were cut on aluminum baseplate, but old mount had too much friction in the fit. Adjusting design for a shank screw with heat set inserts.

nonagaga, 2025-09-29:
Mount was reprinted with captive nuts, fully secured on Computer plate. Pending a review by @space-cat-explorer i think this is a wrap!

closed 2025-09-29 by space-cat-explorer