uf-mil/mechanical #69: New gripper jaws

amcaleer271, 2025-05-01:
The gripper for Sub9 has already been made, though to grasp this year's objects it will need new jaws. These can be 3D printed, but must be strong enough to not break easily while also being able to grab this years objects (a spoon and a cup) reliably.

The grippers should be tested in lab (tabletop and/or in a bucket of water) before going onto the sub.

amcaleer271, 2025-05-20:
Jeremiah is working on this!

DomenickPugliese, 2025-10-05:
Did not have time to work on the gripper this week. My parents came to town and I also had the career fair. I'm planning on getting the gear done next week.

DomenickPugliese, 2025-10-26:
This week I adjusted hole dimensions on the worm gear to the proper dimensions. I couldn't do much besides that since I'm still waiting for the parts to come in the mail. Once those come in I'll print and assemble a gripper prototype.

DomenickPugliese, 2025-10-31:
This week I redesigned the word gear to have more starts so the levers travel far enough to have proper gripper function. I also redesigned the main body to house the servo. Next, I’m going to print everything and design a mounting mechanism for the gripper.

DomenickPugliese, 2025-11-10:
This past week I printed parts for the gripper and tested them. I then made tolerance adjustments to some of the parts and am working on 3d printing them to test them. I am hoping to have the gripper done by the meeting this week.

DomenickPugliese, 2025-11-17:
This past week I tested the worm gear some more. The gear gets stuck on the rough patch from the supports. I'm going to try and print a new worm gear at a different orientation to test if it will work with a more smooth surface.

DomenickPugliese, 2026-02-10:
I forgot to post a report last week but Im just waiting for parts to be printed so I can test them with the servo. If the worm gear doesn't work I'm going to pivot to using a design with only 2 spur gears.

DomenickPugliese, 2026-02-15:
This past week I got all the parts that I needed to test for the gripper printed. I can't do anything else until I'm able to test the gripper with the servo. When the servo board arrives I'll test if the gripper works and then go from there.

RyanHoburg, 2026-02-15:
The board has arrived and is sitting in the mechanisms drawer.

DomenickPugliese, 2026-04-19:
I am still on crutches, so I haven't been able to do too much. I helped Ali so that she could continue the work on the gripper while I couldn't come into the lab. I also redesigned the body of the gripper so the mounting bracket wouldn't snap in half.

closed 2026-08-19 by dmcaleer0131