uf-mil/mechanical #73: SubjuGator floatation

amcaleer271, 2025-05-11:
The sub needs to be positively buoyant by 5% of its weight (per robosub handbook). This helps us, as if the submarine disconnects or loses power, it will naturally float to the surface. To achieve this ,we use non-compressible foam to increase the buoyancy of the sub.

We need to calculate the volume of noncompressible foam to achieve the above level of buoyancy (plus a little extra in case we add anything else in future). We also need a way of securely attaching the foam in a way that can easily and quickly be removed if necessary (no zip ties),

RyanHoburg, 2025-05-23:
[image] This is the current concept. Need some way to secure the clip from rotating so the foam can be secured and not float above the sub.

RyanHoburg, 2025-05-23:
Might also make it longer and use two bolts/screws so the foam doesn't try and rotate, will experiment first.

RyanHoburg, 2025-05-23:
[image] Could maybe get away with this too if we have long enough fasteners.

RyanHoburg, 2025-05-23:
[image] Or this

RyanHoburg, 2025-05-23:
[image] Added holes to hole nuts to make it so the foam isn't a massive pain to install in the first place.

RyanHoburg, 2025-06-12:
All of the R3318 foam that was left in the lab has been mounted to the sub. Several smaller pieces were simply bolted onto the sub in places that will not inconvenience any work that needs to be done to the sub. The majority of the foam was mounted on the top between the carbon tubes in a way that allows us to adjust the floatation in the x-axis (effectively controlling stationary pitch). More foam will be mounted on a threaded rod between the carbon tubes as well. The threaded rod will allow us to control the floatation in the y-axis so we can adjust the floatation as things are added to the sub and have it float evenly.

amcaleer271, 2025-07-30:
Ryan made some awesome mounts for the foam, and we switched some of the foam to a less dense version (the previous foam was rated way deeper than our operating depth).

Closing!

closed 2025-07-30 by amcaleer271