uf-mil/mechanical #80: SubjuGator Task Props

amcaleer271, 2025-05-27:
We already have some props, but need to have all of them before we start testing missions (shooting for that in 3-4 weeks). Details about the props can be found on the [RoboSub Handbook](https://robonation.gitbook.io/robosub-resources)

We need: 

- Prequalification gate
- Prequalification vertical marker
- Start gate
- Slalom channel markers
- Dropper bin
- Torpedo target
- Octagon
- Path markers (2)
- Gripper props

Some of these are already built or just need small modifications, see the respective sub issue for an update on that prop

amcaleer271, 2025-06-10:
The prequal gate and marker are done

amcaleer271, 2025-07-30:
We have props for:
* prequal gate
* nav channel markers (2 of each color)
* start gate
* path markers
* dropper bin

adamhamdan8, 2026-03-30:
The task stuff came out but still not in a rush to get this done because sub is mostly working on nav channel and movement stuff rn.

closed 2026-08-19 by dmcaleer0131