uf-mil/mechanical #91: Thermal Dissipation

amcaleer271, 2025-08-21:
The inside of the computer box gets quite hot under operation, especially if we're using vision. There is no way of getting heat out of the computer box except through conduction through the outer shell. 

We want a way of better extracting heat from the computer box. The main source of heat are the ESCs and the Jetson, so finding a way to better thermally connect these to the outer shell would be great!

Research ways of extracting heat from the computer box, then present your designs to leadership. After that we'll move on to fabrication!

adamhamdan8, 2025-09-19:
Was thinking about this some more and I think we need to really test the necessity of this project. I think we should do some stress testing at one of our pool sessions, and just kinda crank everything to the max and see how the thermals are in the box and see if theres any throttling. Then we can come up with some prototypes for a few different designs and stress test those as well so that we can select the most effective.

I also found some commercial off the shelf options we may want to try, but I gotta confirm that they are for the right jetson. the one we have is mystical and not common apparently

raffimadalian10, 2025-09-21:
Sounds like a plan. I also put together a secondary plan if the OTS option doesn't work if you want to review it. It would involve 3D printing a small duct that can fit over the fan of the Jetson. The hot air would be then directed over the fins mounted on the enclosure wall. I included a rough drawing and some clearance dimensions from the inside of the box. 

[Mil.pdf](https://github.com/user-attachments/files/22453509/Mil.pdf)

[image]
[image]
[image]

raffimadalian10, 2025-09-21:
[image]

RyanHoburg, 2025-09-21:
Before you spend a lot of time designing anything in the computer box, I would get a piece of clay or something soft that can be squeezed between the Jetson and computer box so you can confirm how much space you have. The CAD is not entirely accurate and some components may already be compressed when the lid is closed.

adamhamdan8, 2025-10-16:
Schwartz suggested we look into a compressible thermal foam like thing see if that exists. Electrical will get to figuring out a testing procedure so we can measure temps

raffimadalian10, 2025-10-20:
Had a busy weekend have some midterms this week but will start looking into something compressible to avoid stress on the CPU when closing the lid and come into lab to see if we can get the fan off easily.

raffimadalian10, 2025-11-11:
Sounds good

adamhamdan8, 2025-11-14:
Just as an update, the thermal sensor was not ready last week after testing, going to try and get this done this week if the sensor is properly working

raffimadalian10, 2025-11-17:
Sounds good