uf-mil/mechanical #99: I/O Panels

amcaleer271, 2025-08-25:
We need to create a panel that will allow connections between the pelican boxes on NaviGator. 

The connectors between the boxes will use waterproof connectors, but we need a way of interfacing these connectors with the electronics inside the boxes.

Research and design a method to accomplish this, then present your design and method to leadership!

isanierthm, 2025-09-13:
Researched information and talked with a team lead to better understand the task of creating the panel, but I still need to learn more specifics about NaviGator that will aid me when I begin designing. Next week, I will talk with more members who are working on NaviGator to gather this information.

isanierthm, 2025-09-21:
I talked with the mech leads on NaviGator about the specifications for the panels, and we decided to have two panels bolted onto each box. I took some measurements and made a rough sketch of the outline. I'll CAD an outline next week and talk to electrical to see what holes I need to make in the panels, and then I can decide what size/shape works better.
[image]

isanierthm, 2025-09-28:
I researched a little more this week, just to get a better understanding of the purpose of the panels, as well as trying to brainstorm ideas on how to secure them to the box. I concluded that bolting them to the box would be the best method, so I need to make sure I have enough space on the box to do that with the current dimensions for the panel. Also still waiting on electrical response!

isanierthm, 2025-10-05:
Talked with Jack, and he gave me a general list of the holes needed on the panels, but I'll be meeting with him sometime this week to discuss in more detail.

Here's the list he gave me:
- Motor cable holes 
- Battery in holes
- USB holes for communication with computer box
- Waterproof holes for Coax wire antenna (x2) 
- Ethernet
- Misc power delivery

isanierthm, 2025-11-03:
[image] CAD of the first panel is done! I'll find out how to print it this week, and then we want to do a stability test before we drill any holes.

isanierthm, 2025-11-09:
This week, I didn't get to do a lot because I needed to wait on Jack to start the design for the second panel for the computer box, but we are having a meeting tomorrow. I'll show him the design for the first panel, have it printed, and start the second one this week.

amcaleer271, 2026-01-03:
We routed all of the cables through the boxes in small holes so we could test it before break. With not using it for competition we'll close this issue