uf-mil/mechanical #118: Move depth sensor

amcaleer271, 2025-09-29:
We have identified that when the depth sensor is blasted by water from the thruster positioned next to it, we receive extremely wrong readings, and sometimes the sensor just drops out. This is quite problematic as it makes it difficult for the sub to maintain a constant depth.

We want to move the depth sensor away from the thrusters to get a more accurate and consistent reading. The depth sensor can be put in any pressure vessel as long as there is a way to connect it to the main computer box. Check with the electrical team to determine what kind of cable standard will be necessary.

dmcaleer0131, 2025-10-02:
(10m depth is ~30 psi or 2 bar)

Possible other sensor options:

https://core-sensors.com/sensor-types/submersible-pressure-transducer/cs12-submersible-pressure-transducer/
Rated up to 100 psi 

https://www.sensorsone.com/sea-depth-300ft-5vout-waterproof-swl/
Rated up to 10 bar (145 psi)
G1/4 male din3852 thread
Cable type: PUR

https://www.sensorsone.com/low-cost-rov-10bara-seawater-pressure-sensor/
Rated up to 10 bar (145 psi)
Small DIN plug electrical connection
G1/4 male thread

I don't know exactly what electrical connection is needed yet. Also waiting to hear back on price estimates on sensors. I have been told that price should be under $100.

adamhamdan8, 2026-04-06:
Depth sensor physically moved to the bottom of the sub. Not tested yet, but I will reopen if there is somehow an issue. Steps will be to test in box before we physically reassemble the whole thing in case there is an issue, then fully reassemble.